RobotLocomotion/drake #18991: bazel build //... err

dbdxnuliba, 2023-03-14:
### What happened?

dell@dell:~/ref/drake$ bazel build //...
INFO: Repository spdlog instantiated at:
  /home/dell/ref/drake/WORKSPACE:10:22: in <toplevel>
  /home/dell/ref/drake/tools/workspace/default.bzl:385:29: in add_default_workspace
  /home/dell/ref/drake/tools/workspace/default.bzl:304:26: in add_default_repositories
Repository rule spdlog_repository defined at:
  /home/dell/ref/drake/tools/workspace/spdlog/repository.bzl:29:36: in <toplevel>
ERROR: An error occurred during the fetch of repository 'spdlog':
   Traceback (most recent call last):
	File "/home/dell/ref/drake/tools/workspace/spdlog/repository.bzl", line 27, column 13, in _impl
		fail(error)
Error in fail: error 1 from [/usr/bin/pkg-config, "spdlog"]: 
ERROR: /home/dell/ref/drake/WORKSPACE:10:22: fetching spdlog_repository rule //external:spdlog: Traceback (most recent call last):
	File "/home/dell/ref/drake/tools/workspace/spdlog/repository.bzl", line 27, column 13, in _impl
		fail(error)

### Version

drake-1.13.0

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2023-03-14:
The https://drake.mit.edu/from_source.html instructions have "Mandatory platform-specific instructions" that say to run `sudo ./setup/ubuntu/install_prereqs.sh`.  Did you run that, and did that script finish without any errors?

jwnimmer-tri, 2023-03-14:
Hopefully my suggestion above helps.

I'm closing this as out of scope for the bug database.  Our https://github.com/RobotLocomotion/drake/issues/new/choose requests that "Build or Installation Problems" get posted to Stack Overflow, not bug reports.  If are still having trouble, please post a question there.