RobotLocomotion/drake #19003: Investigate DockerHub runway

jwnimmer-tri, 2023-03-15:
We offer DockerHub images:

https://drake.mit.edu/docker.html

Recently, they have been making things spicy:

- https://news.ycombinator.com/item?id=35154025
   - https://web.docker.com/rs/790-SSB-375/images/privatereposfaq.pdf
- https://news.ycombinator.com/item?id=35166317
   - https://blog.alexellis.io/docker-is-deleting-open-source-images/

Will they light Drake on fire and destroy everyone who uses our binaries in the cloud?  (For example, our tutorials?)

Maybe we should look into https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-container-registry.

At minimum, we should figure out which email address we've signed up with them, and check it for a burnination announcement.

jwnimmer-tri, 2023-03-15:
Xref https://github.com/docker/hub-feedback/issues/2314

jwnimmer-tri, 2023-03-16:
Update posted: https://www.docker.com/blog/we-apologize-we-did-a-terrible-job-announcing-the-end-of-docker-free-teams/

jwnimmer-tri, 2023-03-21:
I proposed a plan on slack. Here's the call to action:

In our DockerHub org, we currently have 4 authorized users (the robot account we actually use, russ, and two 6-years-stale RLG people).  Remove the 3 people, leaving only the bot account.

Then do *Can I migrate to a Personal account?* from https://www.docker.com/blog/we-apologize-we-did-a-terrible-job-announcing-the-end-of-docker-free-teams/ via a support ticket.

BetsyMcPhail, 2023-03-23:
All users except **drakejenkinsbot** have been removed and a support ticket has been opened.

jwnimmer-tri, 2023-03-24:
> **We’re No Longer Sunsetting the Free Team Plan**

-- https://www.docker.com/blog/no-longer-sunsetting-the-free-team-plan/

closed 2023-05-22 by jwnimmer-tri