RobotLocomotion/drake #19034: Improved diagnostic for parsing non-tree-ordered joints

jwnimmer-tri, 2023-03-20:
This error can occur when parsing a URDF:

> RuntimeError: When creating a model, an attempt was made to add two inboard joints to the same body; this is not allowed. One possible cause might be attempting to weld a robot to World somewhere other than its base link; see Drake issue #17429 for discussion and work-arounds, e.g., reversing some joint parent/child directions. Another possible cause might be attempting to form a kinematic loop using joints; to create a loop, consider using a LinearBushingRollPitchYaw instead of a joint

Would it be possible to print out the name of the body that has two inboard joints, and the name of these two inboard joints? I think that would help debugging this error.

jwnimmer-tri, 2023-03-20:
The challenge here is that `class MultibodyTreeTopology` does not know about names, only the integer indices.

Assigning to +@amcastro-tri for suggestions on how to approach a solution.

Shall we add names into topology data structures?  Or pointers to names?

Or provide a callback function to the topology so it can ask about names for error messages?

Or change `add_mobilizer` to use something like [std::expected](https://en.cppreference.com/w/cpp/utility/expected) to _return_ the error message instead of throwing it, so that the caller (MbT) can add more context to the error message?

jwnimmer-tri, 2023-03-20:
That's why I qualified the citation with "something like".

rpoyner-tri, 2023-03-20:
\cc @sherm1 does a case like this have any influence on your proposed mb-tree restructuring work?

sherm1, 2023-03-20:
I am currently (post-vacation that is) reworking the MbP graph analysis code, which will now live at a higher level than the MultibodyTreeTopology stuff and hence have access to MbP objects with friendly names. Hopefully the need for these messages will be reduced since we'll be able to deal with reversed joints and ultimately loops, but I'll make sure it produces helpful messages for cases it can't deal with.

It may also be possible to tweak MultibodyTreeTopology to improve its error messages now but the medium-term plan is for that to disappear altogether when I get the new graph analysis code up.

amcastro-tri, 2023-03-21:
> Shall we add names into topology data structures? 

That is not a bad idea at all. However, Sherm's work will essentially do that in a much cleaner revisited structure that'll allow us to cleanly reason about directed graphs and trees. We could add the names now if a high priority, but my vote would be to wait for Sherm's work to land.