RobotLocomotion/drake #19057: macOS x86_64 releases and CI

jwnimmer-tri, 2023-03-23:
See https://github.com/RobotLocomotion/drake/issues/17210 for background.  Short summary: our macOS x86_64 cloud hosting is Monterey-only and unsupported beyond that, meaning that it will probably die off later this year or next year.

We need to find another way to run x86_64 for macOS -- at minimum, for monthly release packages.  Ideally, for nightly packaging builds as well.  And even most ideally, with nightly CI that actually runs all of the tests.

One option would be GitHub actions.

The fallback option is to take one of our Mac Mini machines, reimage it from scratch, and run it in our own closet to do manual builds on it for monthly release packages.

jwnimmer-tri, 2023-04-12:
As of today at least, GitHub actions [macoOS runners](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-software) also top out at macOS 12.  Hopefully they get macOS 13 in there before 14 is out and Drake drops support for 12.

The (small) runner currently offers:
```
    3-core CPU (x86_64)
    14 GB of RAM
    14 GB of SSD space
```

I'll see if I can get a release build ETA on that hardware profile, to see if it's plausible for us to have a GitHub action for building release binaries there.

jwnimmer-tri, 2023-04-12:
The GitHub Actions free tier seems to be 2000 minutes.  Free builds on macOS machines claim to burn at 10x rate, which means only 200 minutes per month total!

However, given that `drake-external-examples` runs a mac build every night that takes 20-30 minutes, I must be misunderstanding something or we'd be over our limit regularly.

jwnimmer-tri, 2023-04-12:
Huh.  RobotLocomotion is only being billed for https://github.com/RobotLocomotion/drake-jenkins-jobs (which is tiny).  I wonder where the d-e-e billing is going.

jwnimmer-tri, 2023-07-27:
> Aha. I found it on their roadmap:
> https://github.com/github/roadmap/issues/620

The issue was just finished.  GitHub Actions now supports Ventura on x86_64.

jwnimmer-tri, 2024-06-14:
I've used that prototype now to release our wheels for 1.29 and 1.30.  I think it's ready for prime time.  Once we have the https://github.com/RobotLocomotion/drake/issues/21574 flag in Drake, we can work on moving that repo into playbook production.

jwnimmer-tri, 2024-07-25:
And for 1.31, the Drake pin was able to be the exact release tag, no special patches.  So this is basically ready to go into production as part of the https://drake.mit.edu/release_playbook.html.

The goal is to take https://github.com/jwnimmer-tri/drake_github_workflow_prototype/ and move it into RobotLocomotion somewhere, either still as a separate repository, or possibly added to the `drake-ci` repo?

The yaml also needs a few tweaks to make it suitable for the playbook, e.g.:
- Should simple pushing trigger a build (like my WIP has today), or do we want some kind of "staging" page to kick it off?
- Authentication token management (for the SNOPT code).
- Do we want create nightly release candidates?
- ...

jwnimmer-tri, 2024-09-13:
One more new detail...

My prototype uses macOS 13 as its base image.  The Drake release v1.33.0 today was the last one to be based on macOS 13.

The GHA yaml file will need to be changed to target macOS 14 now, to align with the rest of Drake for the next v1.34.0 release.

jwnimmer-tri, 2024-09-13:
For the record, here is the screen where I've added a personal access token for my prototype's GHA to be able to clone snopt:

[image]

mwoehlke-kitware, 2024-10-03:
So, (one of) the first step(s) here seems to be moving your repo to RobotLocomotion. Which means... we need a name.

Some ideas:
- drake-wheel
- drake-build-wheel
- drake-release
- drake-build-release

WDYT?

jwnimmer-tri, 2024-10-03:
It's plausible that this repository will grow to encompass making release binaries beyond just the wheels, so I would probably avoid using "wheel" in the name.  So I think `drake-release` is fine, or `drake-release-builder` is also fine.

jwnimmer-tri, 2024-10-07:
> My prototype uses macOS 13 as its base image. The Drake release v1.33.0 today was the last one to be based on macOS 13. The GHA yaml file will need to be changed to target macOS 14 now, to align with the rest of Drake for the next v1.34.0 release.

According to [here](https://docs.github.com/en/actions/using-github-hosted-runners/using-github-hosted-runners/about-github-hosted-runners#standard-github-hosted-runners-for-public-repositories) there is **no** free runner for macOS 14 on x86. The only option seems to be the `macos-14-large` runner [here](https://docs.github.com/en/actions/using-github-hosted-runners/using-larger-runners/about-larger-runners#about-macos-larger-runners), which requires buying a cloud plan and billing by the minute.

I guess the next step here is to decide whether we want to embrace all of that complexity, or instead just drop this last vestige of support for x86 macs.

jwnimmer-tri, 2024-10-07:
I will take over this issue now.  The remaining work is to document that we no longer support x86 on macOS at all.  I might also publish one final 1.34 wheel next week.

jwnimmer-tri, 2024-10-18:
Earlier this week I released a maxOS x86_64 wheel for Drake v.1.34.0, which will be the last mac wheel for that architecture.

https://drake.mit.edu/release_notes/v1.34.0.html

Nothing more to do here.

closed 2024-10-18 by jwnimmer-tri