RobotLocomotion/drake #19076: MeshcatVisualizer doesn't obey IllustrationProperties ("phong", "diffuse_map")

SeanCurtis-TRI, 2023-03-28:
### What happened?

When specifying a texture for an object via ("phong", "diffuse_map") the texture does *not* get applied to the model in `Meshcat`. It should.

SeanCurtis-TRI, 2023-09-07:
Duplicated and elaborated by #19077.

jwnimmer-tri, 2023-09-07:
This one was about Meshcat.  (The actual 3js scene.)

That one was about Meldis.  (We're failing to even _transmit_ the data.)

SeanCurtis-TRI, 2023-09-14:
A fair point. Both of these, ultimately, are transmission errors. In this case, the texture information simply doesn't get transmitted from `Meshcat`/`MeshcatVisualizer`. `meldis` is lacking input via LCM messages. `MeshcatVisualizer` has the data, but doesn't send it.

SeanCurtis-TRI, 2024-09-27:
BTW When it *works* it looks like this:

MESHCAT:
[image]

The corresponding RenderEngineVtk
[image]

The basic scaffolding (but incomplete) can be found in [this branch](https://github.com/SeanCurtis-TRI/drake/tree/PR_meshcat_texture_primitives).

SeanCurtis-TRI, 2024-09-27:
Note: These images exhibit the issue raised in #20495. The sphere and ellipsoid have been massaged so that they match. At first glance, the boxes may look the same, but they're not. There is no simple operation (like that used on the sphere/ellipsoid) that can be employed on the other shapes to achieve the same end.