RobotLocomotion/drake #19092: [cmake] Use of CMAKE_{COLOR,VERBOSE}_MAKEFILE invalid

svenevs, 2023-03-29:
Surfaced in #19023.

Drake is using unofficial generator-specific entry points to determine the underlying `bazel` command behavior.  As a result, the default behavior between drake's CMake when run using the `Unix Makefiles` generator and any other generator (e.g., `Ninja`) is inconsistent.

- When makefile is used, the defaults are: `color=yes`, `subcommands=no`.
- Any other generator: `color=no`, `subcommands=no`.

One option is to just make them official entry points, `option(CMAKE_VERBOSE_MAKEFILE ...)`.  A slightly better option:

```cmake
set(drake_color_default ON)
if(DEFINED CMAKE_COLOR_MAKEFILE)
  set(drake_color_default "${CMAKE_COLOR_MAKEFILE}")
endif()
option(DRAKE_BUILD_COLOR
  "Enable/Disable color output during build." "${drake_color_default}"
)

set(drake_verbose_default OFF)
if(DEFINED CMAKE_VERBOSE_MAKEFILE)
  set(drake_verbose_default "${CMAKE_VERBOSE_MAKEFILE}")
endif()
option(DRAKE_BUILD_VERBOSE
  "Enable/Disable verbose (bazel subcommands) build output." "${drake_verbose_default}"
)
```

This isn't a bug with CMake, this is a bug with how drake is leaning on certain CMake variables.  And how `if(SOME_VARIABLE_NOT_DEFINED_YET)` behaves to always be falsey, plus the fact that these variables are not defined by default for anything other than the makefile generator.

CC @jwnimmer-tri @mwoehlke-kitware

jwnimmer-tri, 2023-03-29:
I'll take the follow-up of deleting color support entirely (which was added in #10183).  It's not worth any amount of hassle to maintain.

jwnimmer-tri, 2023-03-29:
It seems to me like these are the canonical two we should be supporting instead (if we keep the features):

https://cmake.org/cmake/help/latest/envvar/VERBOSE.html

https://cmake.org/cmake/help/latest/variable/CMAKE_COLOR_DIAGNOSTICS.html

PR #22110 (2024-11-11): [cmake] Remove buggy color and verbosity support
Closes #19092.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22110)
closed 2024-11-11 by jwnimmer-tri