RobotLocomotion/drake #19093: New user "Overview" tutorial

jwnimmer-tri, 2023-03-29:
I think some users (probably many users) would benefit from a very first tutorial that shows a fully-worked example, which uses all of the typical subsystems (simulation, control, planning, rendering, hydroelastic).  Rather than trying to explain everything in detail, it shows all of the pieces in one place in broad strokes, then links to the other bootstrap tutorials for how-to next steps (authoring multibody simulation, dynamical systems, etc.).

More to follow.

JShep-tri, 2023-04-12:
I'm thinking it might be quite convincing to use the manipulation station example here.  It could be quite convincing to a first time user to be able to actually use a robot arm to do some pick and place out of the gates.  I also think that example contains all of the aspects of setting up the sim that we would want a user to see.

It'd be nice to also have a planner and controller aspect, in this way, it would essentially be an extension of https://github.com/RobotLocomotion/drake/blob/master/examples/manipulation_station/joint_teleop.py.

The complexity of the example might be difficult to format into an overview/beginner tutorial, but if broken down in digestible chunks this could be mitigated.

Thoughts @jwnimmer-tri ?

jwnimmer-tri, 2023-04-17:
I like the broad idea.

I'm not sure if literally starting from the current `manipulation_station` code is the best idea due to #14992 pending.  Possibly that porting would be the first step for the tutorial.

I'm also not sure that the "station diagram" approach taken by that code is what we want to show users.  For many use cases, firewalling the dynamics behind a diagram bubble (so that they can swap out robot hardware vs simulated software at the diagram port level) is worse than the simple alternative of just one big diagram without such a firewall.

In short, I think probably "teleop demo with path planning and dynamics simulation" is a good showcase to touch on all of the high points, but we'll probably need to write it from scratch (pulling ideas from existing code, but probably not copy-pasting it).