RobotLocomotion/drake #19094: Workflow for remote drake_models + Bazel + binary Drake

jwnimmer-tri, 2023-03-29:
Some people use a pre-compiled version of Drake with Bazel as their build system, but do not declare `package://drake_models` as a repository external.

For example, the `drake_bazel_installed` workflow, or using `@rules_python//python:pip.bzl` with `pydrake`.

As it stands today, when running bazel tests in this case, the `$TEST_TMPDIR` sandboxing always wins, so all download caching is defeated.  This means tests could be doing a lot of downloading.

We should provide some way for those downstream tests to avoid repeated downloading.

jwnimmer-tri, 2024-02-27:
I'm going to call this a duplicate of #19094.

closed 2024-02-27 by jwnimmer-tri

RussTedrake, 2024-02-28:
nit: your last line has the wrong issue number. (it's marked as a duplicate of itself)

jwnimmer-tri, 2024-02-28:
Oops!  I meant #19823.