RobotLocomotion/drake #19100: Replace `install.py --no_strip` option with native `bazel --strip` option

pathammer, 2023-03-30:
**Is your feature request related to a problem? Please describe.**
We sometimes need to install a debug build of drake in order to debug a program linking to libdrake. It is not clear you need to run `bazel run //:install -- --no_strip` as it is not documented [here](https://drake.mit.edu/bazel.html).

**Describe the solution you'd like**
I think it will be a lot more intuitive for users to run the install target using bazel's built in [strip option](https://bazel.build/docs/user-manual#strip)

**Describe alternatives you've considered**
An alternative would be to add some docs for `bazel run //:install` to the drake homepage  

**Additional context**
 learned about needing to add `--no_strip` on issue #14381

jwnimmer-tri, 2023-03-30:
Is it possible for you to run `cmake && make install` instead of `bazel run //:install`?

Our goal is that the CMake listfile offers a full suite of options.  To this specific example -- by default when `-DCMAKE_BUILD_TYPE=Debug` the project will be installed unstripped (modulo one caveat with SNOPT).

At the moment, `bazel run //:install` is not supposed to be a supported public interface, which is why it's not documented.

If there is a challenge with using CMake install, please share the details.  We could improve the CMake install, or else reconsider whether the bazel install target should be promoted to official support and gain more options.

jwnimmer-tri, 2023-03-31:
One thing we could _probably_ do in the meantime is at least detect with `installer.py` has `strip = True` but the Bazel build had debug symbols enabled, and print a warning during the install in that case (with a pointer to some advice).

richardrl, 2023-12-16:
Isn't the point of building with bazel so you can easily access the tests? So in this sense, it seems people developing on the C++ side might like to see the bazel instructions in the docs.

jwnimmer-tri, 2023-12-16:
> Isn't the point of building with bazel so you can easily access the tests?

That's one benefit, I guess.  There's a lot of reasons people might choose to use Bazel.

> So in this sense, it seems people developing on the C++ side might like to see the bazel instructions in the docs.

When using Bazel, there is _no supported way to install Drake_.  When using Bazel, nothing is ever "installed".  The concept does not exist in the Bazel universe.  When you are developing a downstream project using Drake as a dependency via Bazel, that's the [drake_bazel_external](https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_bazel_external) style and nothing is ever installed.  Things are just built from source, and linked/run as necessary.

If you want to _install_ Drake for use by a downstream project, that's what **CMake** is designed for.  Projects that use this style are for example [drake_cmake_installed](https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed) or [drake_cmake_external](https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_external).

I suppose we could try to document the CMake options more carefully, but our goal was to always use the standard spelling of them, so that we wouldn't need a lot of extra docs.

closed 2026-08-21 by jwnimmer-tri