RobotLocomotion/drake #19106: MultibodyPlant random contexts fall back to zero positions, not defaults

jwnimmer-tri, 2023-03-30:
### What happened?

See [StackOverview](https://stackoverflow.com/questions/75881315/pydrake-montecarlosimulation-pose-of-freebody-being-reset) for the original discovery.

Right now, when I call `SetRandomContext` on a MultibodyPlant, the state I get back is all zeros (except for the '1' in any quaternions).  This happens even if I've previously called `SetDefaultFreeBodyPose` on a free body.

The thing I assumed I would get would be the default joint postures as-if from CreateDefaultContext, for state where I haven't configured any random distribution yet.

### Version

v1.14.0

### What operating system are you using?

Other

### What installation option are you using?

Unknown / Other

jwnimmer-tri, 2023-03-30:
My testing was only with free bodies (quaternion floating joints).  However, I _think_ the same bug probably also applies to all joint types.

jwnimmer-tri, 2023-04-18:
Ping to @xuchenhan-tri for your "MbB bugs worth fixing" panel.

xuchenhan-tri, 2023-04-18:
@jwnimmer-tri do you want to assign a priority label to this issue?

jwnimmer-tri, 2023-04-18:
I don't have any particular urgency one way or the other.

In the regime of "MbP computes the wrong answer" bugs, things like #18682 seem more urgent.  For my taste though, I still think we should fix "computes wrong answer" bugs before working on adding any new features.

PR #19230 (2023-04-25): [multibody] Set random state to default when distribution is missing
Closes #19106.

Currently, default positions for floating bodies are stored separately from the default positions of joint dofs. This is violates single source of truth and is the root cause of #19106.

This PR makes sure that for floating bodies the default pose value requested at MbP::SetDefaultFreeBodyPose() is stored in the floating joint connecting world and the body and MbP::GetDefaultFreeBodyPose() retrieves the default value from the floating joint if it exists. This fixes #19106 as a side effect since now there's sin [... cut]
closed 2023-04-25 by xuchenhan-tri