RobotLocomotion/drake #19113: [visualization] Provide N/m and N*m/m in VisualizationConfig

DamrongGuoy, 2023-03-31:
**Is your feature request related to a problem? Please describe.**
It would shorten my code in half if I can specify scaling factors for visualizing hydroelastic force and moment in VisualizationConfig like this (in Python):
```
    ApplyVisualizationConfig(
        config=VisualizationConfig(
                   publish_period = 1 / 256.0,
                   publish_contacts = True,
                   newtons_per_meter= 2e1,          <====== not available right now
                   newton_meters_per_meter= 1e-1),    <====== not available right now
        builder=builder, meshcat=meshcat)
```

Right now I have to do it in two steps like this:
```
    ApplyVisualizationConfig(
        config=VisualizationConfig(
                   publish_period = 1 / 256.0,
                   publish_contacts = False),
        builder=builder, meshcat=meshcat)

    ContactVisualizer.AddToBuilder(
        builder, plant, meshcat,
        ContactVisualizerParams(
            publish_period= 1.0 / 256.0,
            newtons_per_meter= 2e1,
            newton_meters_per_meter= 1e-1))
```

**Describe the solution you'd like**
Could we add the two parameters to VisualizationConfig (in C++)?
```C++
struct VisualizationConfig {
...
  /** Sets the length scale of the force vectors. */
  double newtons_per_meter{10};

  /** Sets the length scale of the moment vectors. */
  double newton_meters_per_meter{3};
};
```

I will provide a draft PR for consideration.

**Describe alternatives you've considered**
Status quo is ok with me. Perhaps we could document in ApplyVisualizationConfig() that users can still do custom settings afterwards.  At first, I wasn't even sure whether I can add ContactVisualizer after ApplyVisualizationConfig() or not.

**Additional context**
Add any other context or screenshots about the feature request here.

jwnimmer-tri, 2023-04-01:
Are we sure that we _want_ to configure this via API calls?  I haven't used the options myself, but I image there's a chance we would prefer to have a UI element like a slider or control panel to configure them.  If we only have these settings in Params as a stop-gap towards a real solution, I'd rather do the real solution that copy the stop-gap into more Params objects.

DamrongGuoy, 2023-04-04:
I agree that having a UI element would be nice, and I also think an API call is nice too.

At the beginning of setting up the simulation, we don't know the range of magnitude of force and moment vectors. Therefore, having a UI element to adjust the length of these vectors on the fly during playbacks would be very helpful.

After the first author runs the simulation enough times, the author has a better idea of the appropriate force scaling: 1 N/m, 10 N/m, or 100 N/m.  Then, the author can use API calls to save that knowledge into the input deck for the convenience of the next person running the same simulation.

That would be two use cases of UI elements and API calls.

DamrongGuoy, 2023-04-06:
The *all sliders in general* sounds good to me too.

You meant like saving all slider values into a configuration file that we keep with the input deck, right?

By the way, how should I learn about adding sliders?

closed 2026-09-10 by jwnimmer-tri