RobotLocomotion/drake #19135: ComputeSignedDistancePairClosestPoints on a filtered pair

jwnimmer-tri, 2023-04-05:
**Is your feature request related to a problem? Please describe.**

QueryObject.ComputeSignedDistancePairClosestPoints has this to say:

```c++
  /** A variant of ComputeSignedDistancePairwiseClosestPoints() which computes
   the signed distance (and witnesses) between a specific pair of geometries
   indicated by id.
   ...
   @throws std::exception if ... the pair (A, B) has been marked as filtered ... */
  SignedDistancePair<T> ComputeSignedDistancePairClosestPoints(
      GeometryId geometry_id_A, GeometryId geometry_id_B) const;
```

There are cases where it's useful to compute the signed distance, even for objects which cannot collide.  This measurement can inform control algorithms or teleop panels.

**Describe the solution you'd like**

Change `ComputeSignedDistancePairClosestPoints` to ignore collision filters in this function, so that it returns a value this case instead of throwing.

**Describe alternatives you've considered**

Maybe we could make "filtered throws vs filtered returns" a function argument?  I don't see any reason the user ever _wants_ an exception in this case, though.

**Additional context**

[Slack thread](https://app.slack.com/client/T0JNB2DS4/C2WBPQDB7/thread/C2WBPQDB7-1680472087.332559)

jwnimmer-tri, 2023-04-05:
My illustration of a guess at a solution:

```patch
diff --git a/geometry/proximity/distance_to_shape_callback.cc b/geometry/proximity/distance_to_shape_callback.cc
index ed99e48b56..2d85ee7ea9 100644
--- a/geometry/proximity/distance_to_shape_callback.cc
+++ b/geometry/proximity/distance_to_shape_callback.cc
@@ -236,8 +236,9 @@ bool Callback(fcl::CollisionObjectd* object_A_ptr,
   const EncodedData encoding_a(*object_A_ptr);
   const EncodedData encoding_b(*object_B_ptr);
 
-  const bool can_collide = data.collision_filter.CanCollideWith(
-      encoding_a.id(), encoding_b.id());
+  const bool can_collide =
+      (data.collision_filter == nullptr) ||
+      data.collision_filter->CanCollideWith(encoding_a.id(), encoding_b.id());
 
   if (can_collide) {
     // Throw if the geometry-pair isn't supported.
diff --git a/geometry/proximity/distance_to_shape_callback.h b/geometry/proximity/distance_to_shape_callback.h
index 5843ca3ada..43a4ee41ce 100644
--- a/geometry/proximity/distance_to_shape_callback.h
+++ b/geometry/proximity/distance_to_shape_callback.h
@@ -41,6 +41,7 @@ struct CallbackData {
    least as long as the CallbackData instance.
 
    @param collision_filter_in     The collision filter system. Aliased.
+                                  A nullptr means all collisions are allowed.
    @param X_WGs_in                The T-valued poses. Aliased.
    @param max_distance_in         The maximum distance at which a pair is
                                   reported.
@@ -54,13 +55,12 @@ struct CallbackData {
         X_WGs(*X_WGs_in),
         max_distance(max_distance_in),
         nearest_pairs(*nearest_pairs_in) {
-    DRAKE_DEMAND(collision_filter_in != nullptr);
     DRAKE_DEMAND(X_WGs_in != nullptr);
     DRAKE_DEMAND(nearest_pairs_in != nullptr);
   }
 
-  /* The collision filter system.  */
-  const CollisionFilter& collision_filter;
+  /* The collision filter system (if enabled). */
+  const CollisionFilter* collision_filter;
 
   /* The T-valued poses of all geometries.  */
   const std::unordered_map<GeometryId, math::RigidTransform<T>>& X_WGs;
diff --git a/geometry/proximity_engine.cc b/geometry/proximity_engine.cc
index 02989a8b86..b0c5ddf5e5 100644
--- a/geometry/proximity_engine.cc
+++ b/geometry/proximity_engine.cc
@@ -603,7 +603,7 @@ class ProximityEngine<T>::Impl : public ShapeReifier {
     std::vector<SignedDistancePair<T>> witness_pairs;
     double max_distance = std::numeric_limits<double>::infinity();
     // All these quantities are aliased in the callback data.
-    shape_distance::CallbackData<T> data{&collision_filter_, &X_WGs,
+    shape_distance::CallbackData<T> data{nullptr, &X_WGs,
                                          max_distance, &witness_pairs};
     data.request.enable_nearest_points = true;
     data.request.enable_signed_distance = true;
```

DamrongGuoy, 2023-04-06:
I set `priority:medium`. Please let me know if it's needed earlier than that.

jwnimmer-tri, 2023-04-11:
@SeanCurtis-TRI I'd appreciate your feedback here about anything we're missing.  Is there any harm (when a user provides two geometry IDs explicitly) to returning the signed distance, ignoring the filters?

I could imagine that it might be _slow_ sometimes (if we haven't built an OBB tree for this query, because the collision pair was filtered), but it seems like it should always be _sensible_ and _possible_?

SeanCurtis-TRI, 2023-04-11:
I have no concrete recollection of the *why* of this particular feature. It has had this property since the day it was added. I'm going to assume that it's a byproduct of the implementation more than a crafted feature. One *might* make the argument that it keeps the semantics of collision filters consistent throughout the file.

That said, it's perfectly reasonable to want to perform this particular query against proximity geometries that were otherwise filtered for contact or the like. I'd definitely support adding a default parameter for specifying "ignore collision filters". It's the matter of a few minutes to implement and test.

SeanCurtis-TRI, 2023-04-11:
It was easy enough to knock out, so I just threw up the PR.

PR #19176 (2023-04-17): QueryObject allows signed distance between a filtered geometry pair
Historically, requesting signed distance between two geometries that had been filtered would throw. While it might make sense to have the geometries filtered for contact, it might still be convenient to know how far they are. This changes the API to allow the caller to demand that collision filters be ignored.

closes #19135

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19176)
closed 2023-04-17 by SeanCurtis-TRI