RobotLocomotion/drake #19137: Upgrade Externals April 2023

BetsyMcPhail, 2023-04-05:
Follow [Semi-automated monthly upgrades](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/README.md#semi-automated-monthly-upgrades) documentation.

Prior issue: #18914

BetsyMcPhail, 2023-04-05:
@mwoehlke-kitware thanks for the reminder! Here's the issue for April external updates.

mwoehlke-kitware, 2023-04-12:
Notes:
- It wasn't immediately obvious how to obtain the checksums needed to do a manual upgrade of buildifier.
- Per https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/qdldl_internal/README.md, as OSQP has not changed and SCS appears to be working, QDLDL has not been changed.

PR's for the others will be coming...

jwnimmer-tri, 2023-04-12:
> It wasn't immediately obvious how to obtain the checksums needed to do a manual upgrade of buildifier.

If you set the checksum to an incorrect value (e.g., `"0" * 64), the complaint when you run bazel will contain a printout of the correct value.  Alternatively, you can download the buildifier file by hand and run `sha256sum` on it.

mwoehlke-kitware, 2023-04-12:
> If you set the checksum to an incorrect value, the complaint when you run bazel will contain a printout of the correct value.

I assumed that would be the case, but wouldn't that require doing so on every platform that has a different binary?

> Alternatively, you can download the buildifier file by hand and run `sha256sum` on it.

That's what I did for #19189. The main thing that "wasn't immediately obvious" was where to *find* those files. I added a note to the `repository.bzl` that should make it more obvious next time.

jwnimmer-tri, 2023-04-20:
Everything has either been upgraded to delegated to a separate already-open PR.

closed 2023-04-20 by jwnimmer-tri