RobotLocomotion/drake #19142: [visualization] Support animation playback of hydroelastic contact patches in MeshCat

DamrongGuoy, 2023-04-05:
**Is your feature request related to a problem? Please describe.**
It is very helpful in dynamic simulation that we can playback animation faster or slower after the simulation finishes. Currently we can playback hydroelastic force and moment, but we do not support playback animation of contact patches. It is mentioned as a TODO [here.](https://github.com/RobotLocomotion/drake/blob/2409de9bbf69004a0e0471e937454804c4c4dab0/multibody/meshcat/hydroelastic_contact_visualizer.cc#L145)

**Describe the solution you'd like**
Add `time` parameter to Meshcat::SetTriangleColorMesh(..., time) , so we can change [these lines](https://github.com/RobotLocomotion/drake/blob/2409de9bbf69004a0e0471e937454804c4c4dab0/multibody/meshcat/hydroelastic_contact_visualizer.cc#L149-L152) to
```C++
      meshcat_->SetTriangleColorMesh(path + "/contact_surface", item.p_WV,
                                     item.faces, colors, false, time);  // <========Add time
      meshcat_->SetTransform(path + "/contact_surface",
                             RigidTransformd(-item.centroid_W), time);  // <========Add time
```

**Describe alternatives you've considered**

**Additional context**
Right now, the animation playback looks like this.  The contact patch (white circle) of the last time step persists and move around the white cylinder in the playback, which is very confusing.  It's like we have a ghost contact patch moving around.

https://user-images.githubusercontent.com/42557859/230230501-92fcb69b-2e85-4684-8bff-5246f3311c44.mp4

DamrongGuoy, 2023-04-05:
@joemasterjohn @RussTedrake

DamrongGuoy, 2023-05-11:
With @SeanCurtis-TRI 's help, I looked around Drake and [rdeits/meshcat](https://github.com/rdeits/meshcat).  This is what I found:

1. [SetTriangleColorMesh()](https://github.com/RobotLocomotion/drake/blob/ec2ee07bae462a5ef64fba1a12361e691c33f17f/geometry/meshcat.cc#L961-L1006) uses structs to organize data packing of meshes; however, it lacks concepts of times.
2. [SetAnimation()](https://github.com/RobotLocomotion/drake/blob/ec2ee07bae462a5ef64fba1a12361e691c33f17f/geometry/meshcat.cc#L1100-L1182) pack the message in-place for efficiency. It has concepts of times; however, it lacks concepts of meshes.
3. rdeits/meshcat/test/[animation.html](https://github.com/rdeits/meshcat/blob/master/test/animation.html) shows how to animate camera and  boxes. [SetAnimation()](https://github.com/RobotLocomotion/drake/blob/ec2ee07bae462a5ef64fba1a12361e691c33f17f/geometry/meshcat.cc#L1100-L1182) follows the pattern in [animation.html](https://github.com/rdeits/meshcat/blob/master/test/animation.html).
4. rdeits/meshcat/test/[animation2.html](https://github.com/rdeits/meshcat/blob/master/test/animation2.html) shows how to animate Obj meshes.

I think we will need to combine features from the above code together.

There's still an open question of whether we could change the mesh across time.  The [animation2.html](https://github.com/rdeits/meshcat/blob/master/test/animation2.html) uses a fixed mesh.

As I learn more about [meshcat](https://github.com/rdeits/meshcat), I'll have better ideas.

DamrongGuoy, 2023-06-13:
I will update [hydroelastic_basics.ipynb](https://github.com/RobotLocomotion/drake/blob/master/tutorials/hydroelastic_basics.ipynb) when we fix this issue of contact-patch playback.

DamrongGuoy, 2023-12-14:
@dmcconachie-tri said the attached mp4 in the issue description didn't play for him.  I attached the GIF file of the same content here.

Drake_issue_19142.gif
[image]

DamrongGuoy, 2023-12-14:
@dmcconachie-tri also reported the same symptom in Animations Playbak.  The contact patches drifted. They are the very thin sliver of red rectangles, which are hard to see without zooming into the GIF a lot.

[image]
[image]
[image]
[image]

DamrongGuoy, 2024-02-02:
There's a hope #20866 could help. Here's a prototype result from playing back a static HTML record from #20774.
[image]

DamrongGuoy, 2024-02-02:
There is also a better idea than #20866 that Jeremy suggested in https://drakedevelopers.slack.com/archives/C43KX47A9/p1706836298149919?thread_ts=1706810068.483669&cid=C43KX47A9

> to have our `meshcat.cc` and `meshcat_animation.cc` internal logic automatically "transpile" the call sequence of `Meshcat::SetObject` and `Meshcat::SetProperty` commands into a visible/invisible animation sequence, rather than teaching every Meshcat-using visualization class about the trick.

DamrongGuoy, 2024-02-14:
Thank you very much, Jeremy.  I confirm this typo: the third line from below in each block should be `t3` instead of `t1`, right?  Here's the diff:

[image]

[image]

jwnimmer-tri, 2024-02-14:
Yes, I believe you're right.

DamrongGuoy, 2024-02-15:
@jwnimmer-tri, thank you for the idea again.  To help me understand better, I have drafted the code in #20959.  I'll ask some questions in Reviewable.

closed 2026-09-10 by jwnimmer-tri