RobotLocomotion/drake #19149: Verify continuous cache and PR pre merge builds match

svenevs, 2023-04-06:
For pre-merge builds that use the cache server, we should confirm that the pre-merge builds have a corresponding continuous build running that populates the cache.

CC @jwnimmer-tri @BetsyMcPhail .

BetsyMcPhail, 2023-04-25:
| Pre-Merge | Continuous |
| ------------ | ------------ |
| [linux-focal-clang-bazel-experimental-everything-release](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-focal-clang-bazel-experimental-everything-release/)| Missing |
| [linux-focal-clang-bazel-experimental-leak-sanitizer](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-focal-clang-bazel-experimental-leak-sanitizer/) | Missing |
| [linux-focal-clang-bazel-experimental-release](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-focal-clang-bazel-experimental-release/) | [linux-focal-clang-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-release/) |
| [linux-focal-gcc-bazel-experimental-everything-release](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-focal-gcc-bazel-experimental-everything-release/) | [linux-focal-gcc-bazel-continuous-everything-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-bazel-continuous-everything-release/) |
| [linux-focal-gcc-cmake-experimental-release](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-focal-gcc-cmake-experimental-release/) | [linux-focal-gcc-cmake-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-cmake-continuous-release/) |
| [linux-jammy-gcc-bazel-experimental-debug](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-jammy-gcc-bazel-experimental-debug/) |  [linux-jammy-gcc-bazel-continuous-debug](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-debug/) |
| [linux-jammy-gcc-bazel-experimental-release](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-jammy-gcc-bazel-experimental-release/) |  [linux-jammy-gcc-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-release/) |

There is a nightly version of the focal clang everything release  job: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-everything-release/. It takes about 45 minutes.

There is a nightly version of the leak-sanitizer job: https://drake-jenkins.csail.mit.edu/view/LeakSanitizer/job/linux-focal-clang-bazel-nightly-leak-sanitizer/. It takes about 28 minutes. There is also a continuous everything-leak-sanitizer job (https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-everything-leak-sanitizer/) that does not write to the cache.

BetsyMcPhail, 2023-04-25:
That seems like a good plan to me, I will make the updates.

BetsyMcPhail, 2023-04-25:
We confirmed that the caching will already be enabled for the new Continous jobs (https://github.com/RobotLocomotion/drake-ci/blob/main/driver/environment.cmake) so this should just be a jenkins update.

closed 2023-04-27 by svenevs