RobotLocomotion/drake #19151: TrajectorySource::UpdateTrajectory doesn't trigger recalculation of output port values

AdityaBhat-TRI, 2023-04-06:
### What happened?

When calling `UpdateTrajectory()` on a `TrajectorySource`, the cached evaluation of the output port is still considered valid. The values get updated only when the context time gets updated.

FYI @sammy-tri

jwnimmer-tri, 2023-04-06:
@AdityaBhat-TRI I assume the situation where you noticed this came from Python code, is that right?  I think in C++ code, the `const` would have saved us.  At least, keeping around a non-const pointer to a System within a Diagram would have been suspicious.  In Python, there would be nothing to cue us into the risk.

---

Other primitives sources only provide a _per context_ way to update their value:

https://github.com/RobotLocomotion/drake/blob/b57702ef1b9d537bac4b9b59c03b2f1800dc5946/systems/primitives/constant_vector_source.h#L63-L65

The use case in #18434 was to change the value before to creating a context.  I think it's fine to have an API to change the "model value", but we should be very clear about whether a function is changing the model value, or a "Parameters" copy of the model value.

I think possibly the answer here is "working as intended" but we're are missing the code to copy the model value into a parameter, draw the output port from a parameter instead of the member field, and change the parameter within a context.

RussTedrake, 2023-04-10:
I agree that putting it in a parameter instead of a member field would be better.  I'll bow out of the actual solution, though, unless I'm needed.