RobotLocomotion/drake #19153: Consolidated home for visualization components

jwnimmer-tri, 2023-04-06:
It's not very easy to locate our visualization component for any given entity.  I think we should move all of the components into one package so they are easy to browse and understand:

These C++ components should move into `drake/visualization`:
- geometry/drake_visualizer.h
- geometry/drake_visualizer_params.h
- geometry/meshcat_animation.h
- geometry/meshcat.h
- geometry/meshcat_point_cloud_visualizer.h
- geometry/meshcat_types.h
- geometry/meshcat_visualizer.h
- geometry/meshcat_visualizer_params.h
- geometry/meshcat_point_cloud_visualizer.h
- multibody/meshcat/contact_visualizer.h
- multibody/meshcat/contact_visualizer_params.h
- multibody/meshcat/hydroelastic_contact_visualizer.h
- multibody/meshcat/joint_sliders.h
- multibody/meshcat/point_contact_visualizer.h
- multibody/plant/contact_results_to_lcm.h

Likewise their python bindings should move into `pydrake.visualization`.

These are widely used, so it will be three steps:

(1) Move the modules and add forwarding shims to the old names continue to work.  Ensure the shims do not appear in any API reference docs, examples, tutorials, etc.

(2) Introduce deprecation warnings.  Maybe at T+6 months.

(3) Remove the forwarding shims.  Maybe at T+12 months.

DamrongGuoy, 2023-04-06:
It sounds really good to me.  I didn't even know some of those files before. Thanks!

jwnimmer-tri, 2023-04-06:
In the interm, we should probably ensure that all of those classes are tagged into Doxygen:
https://drake.mit.edu/doxygen_cxx/group__visualization.html

jwnimmer-tri, 2024-11-02:
Duplicate of #19786.

closed 2026-09-10 by jwnimmer-tri