RobotLocomotion/drake #19161: question about `cd drake-build    cmake ../drake`

dvogureckiy99, 2023-04-08:
### What happened?

Hi, thank you for this project, I don't know where wrote about this little flaw: in [the instructions about installation from source](https://drake.mit.edu/from_source.html) there is a `cd drake-build    cmake ../drake`, but I think there should be `cd drake-build     cmake ../../drake` in such context. T

Thanks for your attention

### Version

v1.14.0

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using CMake, compiled from source code using Bazel

jwnimmer-tri, 2023-04-08:
The existing instructions for fine for me.  Can you please clarify what you are seeing?  Maybe paste an example of your output.  Here's mine.

```console
jwnimmer@call-cps:~/19161$ git clone https://github.com/RobotLocomotion/drake.git
Cloning into 'drake'...
remote: Enumerating objects: 284994, done.
...
Resolving deltas: 100% (222368/222368), done.
jwnimmer@call-cps:~/19161$ mkdir drake-build
jwnimmer@call-cps:~/19161$ cd drake-build
jwnimmer@call-cps:~/19161/drake-build$ cmake ../drake
-- The C compiler identification is GNU 11.3.0
...
-- Build files have been written to: /home/jwnimmer/19161/drake-build
jwnimmer@call-cps:~/19161/drake-build$ make -j
[ 12%] Creating directories for 'drake_cxx_python'
[ 25%] No download step for 'drake_cxx_python'
[ 37%] No update step for 'drake_cxx_python'
[ 50%] No patch step for 'drake_cxx_python'
[ 62%] Performing configure step for 'drake_cxx_python'
Starting local Bazel server and connecting to it...
```

dvogureckiy99, 2023-04-08:
Sorry, my mistake, I have created directory drake-build inside drake folder.

closed 2023-04-08 by dvogureckiy99