RobotLocomotion/drake #19164: Improve default names for automatic floating base joints

RussTedrake, 2023-04-10:
See extended discussion in [slack](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1681051422016589).

The choice to use e.g. `$world_body1` was made [during code review](https://github.com/RobotLocomotion/drake/pull/18390#pullrequestreview-1205100703), but this choice did not consider all use cases.  People writing optimization problems who reference the joint positions by names in constraints don't want something that verbose.  Python users using `namedview` can't eat the `$` in the name.

Request: the default names should be something more like `{"_"*N}{body_name}` for the smallest N that works. 

Although we don't have proper depreciation, the slack message includes some relevant ideas from @jwnimmer-tri .

Related to #19162.

sherm1, 2023-04-10:
Q: if the system consists of four model instances of the same humanoid (say), we'll add floating joints to the torsos. The proposed scheme makes those joints `torso`, `_torso`, `__torso`, and `___torso` with no clear mapping between name and model instance. Is that OK? We could instead guarantee to add underscores in order of model instance index -- is that worthwhile?

sherm1, 2023-04-10:
A single model instance can include multiple floating bodies -- naming the joints for the bodies as in Russ's proposal seems like the natural approach and least likely to require disambiguation.

sherm1, 2023-04-10:
Ah, sorry. I forgot that we can have the same names in multiple model instances. So the only disambiguation we need is if the model contains both a floating body named "xyz" and an _unrelated_ joint named "xyz" in the same model instance (very unlikely!). In that case we would need to make body xyz's joint be "_xyz" to avoid a joint name conflict.

jwnimmer-tri, 2023-04-10:
Fair enough.  I'm happy to allow whoever implements this to think carefully about the naming choices and back it up with unit tests.

closed 2023-10-02 by sammy-tri