RobotLocomotion/drake #19165: Parsing for SDFormat floating joints

jwnimmer-tri, 2023-04-10:
SDFormat does not offer any standard way to specify a floating joint.  (URDF specifies a joint `type=floating`.)  At minimum, we would like to control the joint names of free bodies (see #19164).

Another important use case is defining a floating joint between a manipuland and an end effector, so that we can `Joint::Lock` it to effectively "weld" it to a gripper.

I think we should add `<drake:joint type="floating">` as a Drake extension to the our SDFormat `<drake:joint>` element:

https://drake.mit.edu/doxygen_cxx/group__multibody__parsing.html#tag_drake_joint

See #13691 for relevant discussion of the URDF parsing.

sherm1, 2023-04-10:
BTW while "floating" should be quaternion_floating to avoid singularities by default, I believe there will be many cases where a user will prefer rpy_floating instead (that will be available when #19065 lands).

jwnimmer-tri, 2023-04-10:
> ... (that will be available when #19065 lands).

While #19065 is an excellent step forward, to be clear it does *not* add parsing support for either SDFormat or URDF.  That would need to be in a a follow-up.  I agree it would be nice, though.