RobotLocomotion/drake #19167: Would be nice to be able to disable deprecation warning messages

edrumwri, 2023-04-10:
We typically want to be able to be aware of the deprecation warnings when pinning to a newer Drake version, file an internal issue to address, then squelch further warnings during our builds. The ability to turn off Drake deprecation warnings would be nice.

jwnimmer-tri, 2023-04-10:
For the moment, I think you could add a patch file to your build as a work-around:

```patch
--- a/common/drake_deprecated.h
+++ b/common/drake_deprecated.h
@@ -4,7 +4,7 @@
 Provides a portable macro for use in generating compile-time warnings for
 use of code that is permitted but discouraged. */
 
-#ifdef DRAKE_DOXYGEN_CXX
+#if 1
 /** Use `DRAKE_DEPRECATED("removal_date", "message")` to discourage use of
 certain APIs.  It can be used on classes, typedefs, variables, non-static data
 members, functions, arguments, enumerations, and template specializations. When
```

That would get the compiler annotations, at least.  The `#pragma ... warning` for file renames would still yell, though.
---

For a longer-term fix, would a preprocessor flag be sufficient?

We could have something like `-DDRAKE_IGNORE_DEPRECATED=1` which (if set by the user prior to including Drake headers) would silence the messages.

jwnimmer-tri, 2023-06-26:
> We could offer a pre-processor option like `-DDRAKE_IGNORE_DEPRECATED=1` which (if set by the user prior to including Drake headers) would silence the messages.

@edrumwri would this be a sufficient fix?

jwnimmer-tri, 2023-07-12:
Closing for lack of response.  We can reopen if there is ever renewed interest.

jwnimmer-tri, 2024-03-01:
We decided that we need this inside TRI.

The goal will be flag(s) that control all of:
- The C++ deprecation warnings in drake.
- The "warn once" logger warnings in drake.
- The Python deprecation warnings in pydrake.

jwnimmer-tri, 2024-09-05:
Self reminder: also fix https://github.com/RobotLocomotion/drake-external-examples/issues/159.