RobotLocomotion/drake #19188: RenderEngineGl supports MTL files

zachfang, 2023-04-12:
Currently, `RenderEngineGl` **can** load MTL files via `tiny_obj_loader`, but the code ignores it to be consistent with other RenderEngine implementations.

As the first pass to support MTL files in RenderEngineGl, we will only accept an MTL file with one texture, i.e., a phong texture or `map_Kd`, and ignore if an MTL file has multiple textures with a warning.

jwnimmer-tri, 2023-06-26:
See https://github.com/RobotLocomotion/drake/issues/18844#issuecomment-1602137932 for some more details.

@SeanCurtis-TRI @zachfang we should probably use this ticket to discuss GL details, instead of the Epic ticket?

zachfang, 2023-06-26:
Sounds good. I used that comment to capture the status for all three RenderEnigines and agreed we should have any follow-up Gl-specific discussions here.

closed 2023-07-13 by zachfang