RobotLocomotion/drake #19202: Speed up Q @ X in Python

jwnimmer-tri, 2023-04-15:
See the `MatrixInnerProduct` benchmark in #18870.

Multiplying a double matrix by a Variable matrix should be fast.  At the moment, especially in Python, it is quite slow.

jwnimmer-tri, 2023-04-15:
WIP at https://github.com/jwnimmer-tri/drake/tree/matmul-double-variable

cohnt, 2023-04-18:
Perhaps connected to #10900?

The above WIP has excellent performance with the above benchmark -- I'm seeing the following results:
```
2023-04-18T13:17:09-04:00
Running ./benchmark_polynomial
Run on (20 X 4900 MHz CPU s)
CPU Caches:
  L1 Data 48 KiB (x10)
  L1 Instruction 32 KiB (x10)
  L2 Unified 1280 KiB (x10)
  L3 Unified 24576 KiB (x1)
Load Average: 3.11, 7.75, 12.87
***WARNING*** CPU scaling is enabled, the benchmark real time measurements may be noisy and will incur extra overhead.
-----------------------------------------------------------------------------
Benchmark                                   Time             CPU   Iterations
-----------------------------------------------------------------------------
BenchmarkPolynomialEvaluatePartial       1277 us         1277 us          591
BenchmarkMatrixInnerProduct/10/0        0.000 s         0.000 s          7394
BenchmarkMatrixInnerProduct/50/0        0.013 s         0.013 s            55
BenchmarkMatrixInnerProduct/100/0       0.105 s         0.105 s             6
BenchmarkMatrixInnerProduct/200/0       0.958 s         0.958 s             1
BenchmarkMatrixInnerProduct/10/1        0.000 s         0.000 s         37974
BenchmarkMatrixInnerProduct/50/1        0.001 s         0.001 s          1018
BenchmarkMatrixInnerProduct/100/1       0.004 s         0.004 s           195
BenchmarkMatrixInnerProduct/200/1       0.022 s         0.022 s            32
Gemm/50                                  17.1 ms         17.1 ms           41
```
However, I'm not seeing a significant speedup in Python. The following code (which was the example that lead to the discovery of this bug) is still very slow.
```
import numpy as np
from pydrake.all import MathematicalProgram, Solve
import pydrake.solvers

n = 200
np.random.seed(43)
Q = np.random.randn(n,n)

prog = MathematicalProgram()
X = prog.NewSymmetricContinuousVariables(n)
for i in range(n):
	prog.AddConstraint(X[i,i] == 1)
prog.AddPositiveSemidefiniteConstraint(X)

print("Finished adding constraints.")
print("Computing objective part 1...")
tmp1 = Q @ X
print("Computing objective part 2...")
tmp2 = np.trace(tmp1)
print("Computing objective part 3...")
prog.AddLinearCost(tmp2)
print("Solving...")

MosekSolver = pydrake.solvers.MosekSolver()
ScsSolver = pydrake.solvers.ScsSolver()

solver_options = pydrake.solvers.SolverOptions()
solver_options.SetOption(pydrake.solvers.CommonSolverOption.kPrintToConsole, 1)
prog.SetSolverOptions(solver_options)

# result = MosekSolver.Solve(prog)
result = ScsSolver.Solve(prog)
```
In particular, the `tmp1 = Q @ X` line takes about 40 seconds.

jwnimmer-tri, 2023-04-18:
Can you change your code to `tmp1 = pydrake.sym.matmul(Q, X)` instead of `@`?  The WIP doesn't (and probably won't be able to) supplant `__matmul__`, so `@` will still be slow.

cohnt, 2023-04-18:
Using `tmp1 = pydrake.symbolic.matmul(Q, X)` is extremely fast!

Is there any functionality the original `__matmul__` supplies that this new version doesn't? Or is there an implementation reason why `@` can't call the symbolic matmul? I think this would be an easy way for people to get tripped up when writing SDPs.

jwnimmer-tri, 2023-04-18:
> Is there any functionality the original __matmul__ supplies that this new version doesn't?

I _think_ they should be the same.  At least, by the time the PR is finished and tested, they should be equivalent.  Possibly the WIP has some missing pieces.

> Is there an implementation reason why @ can't call the symbolic matmul?

The challenge with making `@` itself work involves numpy compatibility and what kind of C ABI hooks it offers for us.  It's possible that for a sufficiently new version of numpy found at runtime, we'd be able to do the fast thing automatically.  I will almost certainly defer that to a follow-up PR though, to make the `sym.matmul` available as quickly as possible.  My recollection is that current numpy has sufficient hooks for operations like `max` and `less_equal` but not yet for matmul.  Possibly the very newest has a solution, it's been a few months since I looked.

> I think this would be an easy way for people to get tripped up when writing SDPs.

Yes, for sure.  For now, we'll need to rely on good examples and warnings in the documentation.

cohnt, 2023-04-27:
I've been using your WIP fixes for several weeks now, and they seem to be working really well. Any ideas on when they'll be ready to pull into master? Efficient multiplication is essential to setting up large SDPs, and it's really hard to do right now in Drake without your fixes.

jwnimmer-tri, 2023-04-27:
@cohnt Could you check whether #19298 provides some speed-up?  It will probably be less of a bump than the full WIP branch, but I'm hoping it's still a pretty helpful improvement.  In any case it needs to land before the rest of the WIP.

Note that I've moved the function from `symbolic.matmul` to `math.matmul` to support autodiff as well.

cohnt, 2023-04-27:
I got a small speedup, but it still wasn't huge. With a 200x200 matrix, `Q @ X` takes 38.976608 seconds, and `pydrake.math.matmul(Q, X)` takes 30.053864 seconds.

I used the following script for timing
```
import numpy as np
from pydrake.all import MathematicalProgram, Solve
import pydrake.solvers
import pydrake.math
import time

n = 200
np.random.seed(43)
Q = np.random.randn(n,n)

prog = MathematicalProgram()
X = prog.NewSymmetricContinuousVariables(n)
for i in range(n):
	prog.AddConstraint(X[i,i] == 1)
prog.AddPositiveSemidefiniteConstraint(X)

t1 = time.time()
tmp1 = Q @ X
print("Old: %f" % (time.time()-t1))
t1 = time.time()
tmp1 = pydrake.math.matmul(Q, X)
print("New: %f" % (time.time()-t1))
tmp2 = np.trace(tmp1)
prog.AddLinearCost(tmp2)

MosekSolver = pydrake.solvers.MosekSolver()
ScsSolver = pydrake.solvers.ScsSolver()

solver_options = pydrake.solvers.SolverOptions()
solver_options.SetOption(pydrake.solvers.CommonSolverOption.kPrintToConsole, 1)
prog.SetSolverOptions(solver_options)

# result = MosekSolver.Solve(prog)
result = ScsSolver.Solve(prog)
```

jwnimmer-tri, 2023-04-27:
Okay great, thanks!  I'll work on getting this first step landed first (cutting out the 8s of function call overhead), and then fixing the algorithm second (to be only O(mn) instead of O(m²n²) or whatever horror it is now), to decimate the remaining cost.