RobotLocomotion/drake #19235: Distance Link and Revolute-Spherical Joint/Constraint

MarvinSt, 2023-04-20:
Feature Request:
To make it convenient to model automotive suspension linkages, it would be nice to have simple joints which impose certain constraints, such as a distance constraint (to model a massless rod with two spherical joints) or a combined revolute/spherical joint. Project Chrono offers such types of linkages and they can be used interchangeably with the other more conventional type of joints. (see: https://api.projectchrono.org/8.0.0/links.html)

Solution:
Support similar types of joints as in Project Chrono for cases where the mass/inertia of intermediate bodies can be neglected and the system can be simplified using combination joints.

Alternatives:
N/A

Screenshots:

Consider the system (taken from Project Chrono documentation)
[image]

Which can be modelled using distance constraints (if the linkage bodies can be neglected)
[image]

Or if the bodies should be modelled, using individual or combined (control arm) bodies:
[image]

EricCousineau-TRI, 2023-04-20:
Assigning @amcastro-tri for disposition

MarvinSt, 2023-04-20:
This is also available in Simbody and can be implemented as follows:
```cpp
auto dist = (chassis_link_pos - upright_link_pos).norm();
// Option 1:
auto link = Constraint::SphereOnSphereContact(chassis, chassis_link_pos - chassis_pos, dist / 2.0, upright, upright_link_pos - upright_pos, dist / 2.0, false);
// Option 2:
auto link = Constraint::Rod(chassis, chassis_link_pos - chassis_pos, upright, upright_link_pos - upright_pos, dist);
```

sherm1, 2023-04-20:
I think the distance constraint is already available when using Drake's SAP solver. Is there an example somewhere?

amcastro-tri, 2023-05-01:
Yes, with the SAP solver we can already add distance constraints. 
Documentaiton to the API is here in [`MultibodyPlant::AddDistanceConstraint()`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a8f29f6aa01dd68ee2babe09ace936ebf). Python bindings are available as well.

For spherical joints constraints, which we call ball constrains, these are also supported in SAP.
The API with docs is here in [`MultibodyPlant::AddBallConstraint()`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a0572ce9897e3727ccdfd5fbb34e6512a).

Are these what you need @MarvinSt? should we close the issue or maybe update it to reflect the missing feature?

MarvinSt, 2023-05-01:
Can these be used via Python or only via C++? I had been using Python so far, hence why I opened this ticket.

MarvinSt, 2023-05-01:
I think we can close it, it was not obvious to me how to add constraints via the Python bindings. Perhaps a minimum example would be valuable, but it seems like the feature exists already, my apologies.

amcastro-tri, 2023-05-01:
no problem at all!. The unit tests (though in C++) show how to call these APIs. What did you think about the API docs? are those enough for you to get started? or how do you think we could improve them?

MarvinSt, 2023-05-01:
> no problem at all!. The unit tests (though in C++) show how to call these APIs. What did you think about the API docs? are those enough for you to get started? or how do you think we could improve them?

From my perspective, the docs were quite extensive in terms of background info, but difficult to actually get up and running (esp. compared to Chrono and Simbody for example). Most examples are steering towards creating models from .udf files and for the Python bindings, I indeed mostly tried to transcribe the C++ examples.
It was not hard per-see, it just required a bit more work to get up and running. Where I got stuck was with kinematic loops and constraints. 

I am currently doing the following:
* setup the diagram builder with multi body plant scene graph
```
builder = DiagramBuilder()
plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step=sim_time_step)
parser = Parser(plant)
```
create a model instance
```
vehicle_model = plant.AddModelInstance("vehicle_model")
```
add a bunch of bodies, joints and visualizers...

finalise plant and setup the mesh cat visualizer
```
visualizer = MeshcatVisualizer.AddToBuilder(
    builder, scene_graph, meshcat, MeshcatVisualizerParams(role=Role.kPerception, prefix="visual"))
diagram = builder.Build()
```

but I don't have the ability to do
plant.AddDistanceConstraint() or any constraint for that matter