RobotLocomotion/drake #19236: Tabular Spring and Damper

MarvinSt, 2023-04-20:
Feature Request
Convenient components to model non-linear spring damper systems, for example:
- Non-linear fluid/gas dampers with variable characteristics in bump and rebound (i.e. as described here https://www.extremeshox.com/blog-post/tech-lab-episode-2-origin-damping-forces/)
- Elastic end-stops with engagement gap
- Series springs or progressive springs

Solution
The current linear spring damper element could be extended to accept lookup tables for the force characteristics. Additionally, it would be useful to implement besides the free-length an engagement gap (travel distance after which the spring engages) and a max travel limit (useful for multi-rate series spring with a preload and gap stop).

It would be convenient if multiple force characteristics can be attached to the same spring/damper element (i.e. preloaded spring with a tabular end-stop with a specific engagement gap), without the need to create multiple elements. This would allow one to conveniently attach an end-stop characteristic to an existing spring and adjust the engagement gap without needing to recompute the main spring characteristic.

Ideas for configurable parameters:
* End-stop engagement gap
* Spring compression limit (for series springs, no compression / constant force beyond this limit)
* Option for spring force only under compression (for free springs not fixed to the ends)
* Spring curve (or stiffness coefficient)
* Damper curve (or damping coefficient)
* End-stop curve (or cubic coefficients) (can be on both ends of the travel limit)

Alternatives
N/A

Screenshots:
[image]

[image]

[image]

EricCousineau-TRI, 2023-04-20:
Assigning @amcastro-tri for disposition

FWIW: You should be able to prototype and model this in a continuous-time plant. Once you have a concrete prototype, you can then try out your desired workflow (control? contact-rich manip?), and see how critical that model is to your intended use.

For example, I and others are testing simple stiction models for control to ensure the workflow is vetted, e.g. #18932

MarvinSt, 2023-04-25:
How about something like this (example implemented using Simbody): 
https://github.com/MarvinSt/vehicle-model/blob/main/cpp/springdamper.hpp

I'm not that skilled with C++, but the idea is to have a combined spring damper element with for example 3 lookup tables:
- Spring characteristic (force = table(displacement - x0))
- Damper characteristic with optional static preload (force = table(velocity) + f0)
- Bump stop (end stop) table (force = table(displacement - x0 - x_bump))

```cpp
    Real eval_force(Real d, Real v) const
    {
        Real stretch = d - m_x0;                               // + -> tension, - -> compression
        auto frcSpring = m_spring.eval(-(stretch));            // - -> tension, + -> compression
        auto frcBump = m_bumpstop.eval(-(stretch - m_x_bump)); // - -> tension, + -> compression
        auto frcDamper = m_damper.eval(v);

        return -(frcSpring + frcBump + m_f0); // + -> tension, - -> compression
    }
```
For the lookup table, we can use any preferred implementation (dynamically sized or statically sized, linear or quadratic interpolation, extrapolation yes/no, etc.)

It is of course also nice to think about control inputs for things like active dampers, but that will probably become too bespoke, but it perhaps an external force or damper scaling factor could be considered.

Let me know if you think this is something useful to include in the base lib.

Another reference of a pretty extensive spring damper implementation: 
https://github.com/projectchrono/chrono/blob/main/src/chrono_vehicle/ChSubsysDefs.cpp

sherm1, 2023-04-25:
Hi, Marvin. It is already _possible_ (not necessarily easy!) to write a custom force element in Drake. Please see the [door hinge](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_door_hinge.html) for an example. This was written to model a dishwasher door that has highly nonlinear latching behavior at various positions along its travel. The [header](https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/door_hinge.h) and [implementation](https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/door_hinge.cc) are here.

I am not sure whether you can make similar ForceElement extensions directly from PyDrake -- anyone know?

jwnimmer-tri, 2023-04-25:
> I am not sure whether you can make similar ForceElement extensions directly from PyDrake -- anyone know?

No, the python bindings for `ForceElement` do not allow for subclassing in Python yet; only allow operating on the existing types (LinearSpringDamper, etc) is bound.

Finishing #13258 is probably a prerequisite for adding bindings that would allow for subclassing.

MarvinSt, 2023-04-25:
Thanks for the feedback, I've looked at the door hinge example and that would certainly be one way to do it, but also you stated it is rather low level and not for example supported via the .sdf route. Exposing the Force Elements via Python is certainly be useful, but it would be convenient from an end-user perspective to be able to specify non-linear characteristics natively.

Chrono solves this quite nicely, the user can provide a callback function to overwrite the default linear translational  characteristics, which has the following signature:
```
double ForceFunctor::evaluate(double time,
                                      double rest_length,
                                      double length,
                                      double vel,
                                      const ChLinkTSDA& link)
```
They have an extension with utilities targeted towards vehicle modelling where they provide various useful implementations which can be used directly by the end users (i.e. degressive, tabular, bump stops, etc.). 

I'm not sure if this would be an option for the Drake core library as well? Or is this better left up to the end user to implement? I could try to work on a prototype, but I have not yet touched the C++ API so it will take a while.

sherm1, 2023-04-25:
BTW Drake is focused on robotics so in our own development we prioritize features that are of use to roboticists. The door hinge was part of a project to have a robot load a dishwasher!

dvogureckiy99, 2023-05-11:
@MarvinSt Please contact me [dvogureckiy99@gmail.com](dvogureckiy99@gmail.com), I want to discuss with you idea: make a bridge between `Chrono` and `Drake`, if you are interested in it. Our team nowadays working on the [`Rostok` open source Python framework for generative design of linkage mechanisms for robotic purposes](https://github.com/aimclub/rostok), we successfully using `PyChrono` physical engine and we have accumulated enough experience in this