RobotLocomotion/drake #19286: Update to jenkins-job-builder 5.x

svenevs, 2023-04-26:
5.x got rid of "defaults-view":

```
/__w/drake-jenkins-jobs/drake-jenkins-jobs/views.yaml:32:3: Unknown topmost
element type : 'defaults-view'; known are: defaults,job,job-template,
job-group,view,view-template,view-group,project,parameter,property,builder,
wrapper,trigger,publisher,scm,pipeline-scm,reporter.
```

For now we just pinned requirements to 4.3.0, we need to look into what has changed about views generation.  It could be that we could maybe finally get the table view for continuous production and friends?

CC @jwnimmer-tri @BetsyMcPhail

tyler-yankee, 2025-01-24:
There are several changes that will need to be made to upgrade to the latest, but I believe most are syntactical and the effort should not be too significant. This [document](https://docs.google.com/document/d/1aCmaBs8L5HC25ldWPNAXtb2jiVUbTVyF65HGuhME_yM/edit?usp=sharing) contains notes from the Changelog regarding what will need to be done, and some other useful links.

BetsyMcPhail, 2025-03-26:
@tyler-yankee are we ready to close this issue now?

tyler-yankee, 2025-03-26:
Yes, closed as completed via [#160](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/160) and [#163](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/163).

closed 2025-03-26 by tyler-yankee