RobotLocomotion/drake #19295: Toward removing robotlocomotion/director tap

svenevs, 2023-04-27:
VTK and clang-format are the only things that come from https://github.com/RobotLocomotion/homebrew-director

```console
$ grep -ri robot setup/mac/
setup/mac/binary_distribution/Brewfile:4:tap 'robotlocomotion/director'
setup/mac/binary_distribution/Brewfile:27:brew 'robotlocomotion/director/vtk@9.1.0'
setup/mac/source_distribution/Brewfile:4:tap 'robotlocomotion/director'
setup/mac/source_distribution/Brewfile:7:brew 'robotlocomotion/director/clang-format@12'
```

- [ ] Remove vtk (via #16502 and tarball downloads rather than `brew` installs)
- [ ] Make a decision on what to do about [`clang-format@12` formula](https://github.com/RobotLocomotion/homebrew-director/blob/master/Formula/clang-format%4012.rb)

When building locally using `brew install -v --build-from-source ./clang-format@12.rb`, downloading and staging take a bit, and there are (at least with clang 12) 795 objects being compiled.  AFAICT this gets compiled in serial

- mac mini arm64: ~2 minutes 12 seconds.
- older x86 mac: ~6 minutes 6 seconds.

Seems possibly reasonable to incorporate this into drake's workspace if we so desired.  Then again, this is notably a lot more time than just downloading our precompiled bottle.  There aren't many dependencies for the formula, and as a result we rarely need to rebottle it (haven't needed to since we switched to 12 and did the first one).

CC @BetsyMcPhail @jwnimmer-tri

jwnimmer-tri, 2023-11-30:
Also another WIP [here](https://github.com/jwnimmer-tri/drake/commits/clang-python-bindings) is relevant.  Both should probably happen together.

jwnimmer-tri, 2024-05-09:
Fixed in #21339.

closed 2024-05-09 by jwnimmer-tri