RobotLocomotion/drake #19308: GCSTrajectoryOptimization can reproduce the examples from the paper

RussTedrake, 2023-04-29:
https://arxiv.org/abs/2205.04422

@wrangelvid is making great progress!  We agreed to make this issue to track the status.

Still TODO:
- [ ] Time costs (#19301)
- [ ] Velocity constraints
- [ ] ...

@wrangelvid -- please go ahead and make/update your list here.

RussTedrake, 2024-06-22:
Update: I've ported @wrangelvid 's latest and greatest uav example into the underactuated repo, where I will continue to nurture/maintain it.  https://github.com/RussTedrake/underactuated/blob/master/book/trajopt/gcs_quadrotor.ipynb .  We're just missing the energy costs (really derivative costs) now to implement the minimum-snap objective.  Great work @wrangelvid ! 🎉

RussTedrake, 2024-08-21:
#21810 has landed.  @wrangelvid ... shall we consider this complete and close the issue?

wrangelvid, 2024-08-21:
> #21810 has landed.  @wrangelvid ... shall we consider this complete and close the issue?

Awesome. Looks good to me!

closed 2024-08-21 by wrangelvid