RobotLocomotion/drake #19322: Towards making SAP our default discrete solver

amcastro-tri, 2023-05-01:
This Epic tracks the development of code that, after our recent research, we know how to write. We believe these are the last missing bits towards making SAP the default solver in Drake.

1. Incorporate latest research on convex H&C with regularized friction, #19320. This fixes #18241.
2. Infrastructure for joints to report multibody forces, towards #19068.
2. Report proper reaction forces for arbitrary constraints. Fixes #19068.
3. Implicit PD controllers, #19067.
4. Joint stiction #18932

aminya, 2023-09-13:
One issue with the SapSolver is the `d2ellR_dalpha2` range failure. This line sometimes fails for me, but I cannot gain insight into the reasons.

https://github.com/RobotLocomotion/drake/blob/c482ed7f6222958eb1f2904db9bbbd22e6eedf9f/multibody/contact_solvers/sap/sap_solver.cc#L341

amcastro-tri, 2023-09-25:
Hi @aminya, we actually run into this issue ourselves last week.
The underlying reason was not the solver really but there was garbage (large values, order of 1e300) in the inputs. Since SAP is the only place where numerics is checked, that's were the code first barks. 
I wonder if you had a similar experience. In my dev branch I put a check for those large values, probably worth pushing into master if a common problem. WDYT?

aminya, 2023-09-26:
@amcastro-tri I verified that there are no garbage or large values in the inputs. I disabled the three checks to see what happens. But it fails at this line randomly:

```cpp
newton_with_bisection/drake/multibody/contact_solvers/newton_with_bisection.h:31 in Bracket(): condition 'has_different_sign(f_lower, f_upper)' failed.
```

This is random and it doesn't happen all the time. It's very hard to have a reproducible setup.  

The only reason I am using the SAP solver now is the mimic joint support. Is there an alternative solution for me?

closed 2024-12-17 by jwnimmer-tri