RobotLocomotion/drake #19330: Xcode upgrade

jwnimmer-tri, 2023-05-02:
See https://developer.apple.com/support/xcode/ -- we are running 14.0 but 14.2 is available.  We should upgrade CI to the latest available release, and update the website docs to reflect the CI version.

Note that 14.3 is also available but seems to not support Monterey.  It would be OK to use 14.3 on the Ventura images, i.e., different Xcode versions per OS version is fine.

I believe there's at least one build error with the newer Xcode.

See also #18327.

svenevs, 2023-06-30:
Due to limitations, the arm64-monterey image is stuck at macOS 12.6.3.  The only stock image available is 12.2.  For now, we are unable to update past this version of macOS.  This is orka specific.

svenevs, 2023-06-30:
And the status has been confirmed, we cannot upgrade macOS monterey arm64 past Xcode 14.2, 14.3 and onward require a later macOS.  Ventura update is getting finalized to macOS 13.4.1 and Xcode 14.3.1.

Last one to square away is x86 (vsphere) which is a bit outdated, we anticipate being able to get that up to the latest Xcode version (14.3.1).

svenevs, 2023-07-03:
Final status as of 2023-07-03: ticket is resolved as much as can be.  There are no docs updates AFAICT, we stopped including minor / patch versions in the docs (they just say "Xcode 14").

## x86 (vSphere)

### Monterey

```console
$ sw_vers -productVersion
12.6.7

$ xcodebuild -version
Xcode 14.2
Build version 14C18

$ clang --version
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: x86_64-apple-darkwin21.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xtoolchain/usr/bin
```

_Summary_: everything is working as expected.  We do not anticipate needing to upgrade these images anytime in the near future.  Xcode 14.2 is the latest version.

Packaging build works which is the one we really care about: https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey%20Unprovisioned/job/mac-x86-monterey-unprovisioned-clang-bazel-experimental-packaging/5/consoleFull

Provisioned: https://drake-jenkins.csail.mit.edu/view/Mac%20Monterey/job/mac-x86-monterey-clang-bazel-experimental-release/68/consoleFull

## arm64 (orka)

### Monterey

```console
$ sw_vers -productVersion
12.6.3

$ xcodebuild -version
Xcode 14.2
Build version 14C18

$ clang --version
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: arm64-apple-darwin21.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

_Summary_: we do not anticipate updating the xcode base image anytime soon, there are issues updating to 12.6.7 on the backend and there is no apparent need.  Xcode 14.3+ seem to dissolve monterey support.  There are additional details being sorted out on the backend that do not relate to the xcode upgrade that may result in an updated monterey.  Or ideally, we drop monterey before needing to do that.

### Ventura

```console
$ sw_vers -productVersion
13.4.1

$ xcodebuild -version
Xcode 14.3.1
Build version 14E300c

$ clang --version
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin22.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

_Summary_: we're on the "latest and greatest" of everything.  Ideally, we don't do anything on the base image (we can update unprovisioned / provisioned quickly) until after we want to add a new macOS.

----------------------

All that said Xcode 15 will also be inbound so I guess the adventures will continue.  However, I'm of the opinion that we should delay that task.  If users find issues then we'll deal with it.

closed 2023-07-03 by svenevs