RobotLocomotion/drake #19338: Including SCIP into drake

FranekStark, 2023-05-03:
SCIP is a fast MI solver. In the past, It was not included into drake, because of license issues. (https://github.com/RobotLocomotion/drake/issues/12997#issuecomment-608827345, https://github.com/RobotLocomotion/drake/issues/3367#issue-175333654)

However, recently the license has been changed to APACHE2.0 (https://scipopt.org/index.php#license). Maybe it would be worth it to consider adding, SCIP as a solver to have a nice (non-commerical) alternative to GUROBI and MOSEK.

RussTedrake, 2023-05-03:
Fantastic news!  Thank you for letting us know.  We'll definitely take another careful look.

jwnimmer-tri, 2023-10-12:
I'll let others respond with any solver programming tips, but helping with the build system part of this task would be on my plate.  One question to help get started on that front...

Looking at the [CMakeLists](https://github.com/scipopt/scip/blob/master/CMakeLists.txt) for SCIP, I see that there are some options for turning on/off various dependencies or options.  Do we know which options we would want (or need) to have ON for Drake?  In particular, if the SCIP library requires other helper libraries then we would also need to check the license terms for those libraries.  If you can provide a list of minimum required options to enable, I can help check those licenses.

jwnimmer-tri, 2023-10-12:
On a related note, I also did a quick check of helper libraries that are copied into the SCIP codebase  directly.  FYI It looks like for Drake's use of SCIP we would use either `SYM=nauty` or `SYM=snauty` mode, since the other option (`bliss`) is licensed as LGPL-3.0.

AlexandreAmice, 2024-02-08:
I think this would be a great thing to include in Drake. Currently, we don't really have an open-source MIP solver which feels very limiting. For example `MaxCliqueViaMipSolver` only works if the user can install `Mosek` or `Gurobi`. This will also limit the useability of the forthcoming `IrisInConfigurationSpaceFromCliqueCover` feature.

RussTedrake, 2024-04-22:
I'm also very excited about this.  Having an open-source MISOCP solver and having any MISDP solver would be fantastic contributions (we would try them immediately in our work on Graph of Convex Sets).