RobotLocomotion/drake #19340: Buildcop email template changes URL not isolated to failed build

svenevs, 2023-05-03:
You get an email like

```
Subject: [drake-developers] Jenkins build is unstable: linux-jammy-gcc-bazel-continuous-debug #329

Body:
See <https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/329/display/redirect?page=changes>
```

Redirect page does not isolate you to the changes associated with the build, though which (at least today) can be more confusing when there's failed builds stacked up coming in at different times (e.g., the fix has already landed):

https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/329/display/redirect?page=changes => https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/changes

There is a url https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/329/changes which I think is what we want

Fix in yaml side is easy, relevant section

```
emailext (
    subject: "${{subject}}: ${{env.JOB_NAME}} #${{env.BUILD_NUMBER}}",
    body: "See <${{env.BUILD_URL}}display/redirect?page=changes>",
    to: '$DEFAULT_RECIPIENTS',
)
```

`${{env.BUILD_URL}}display/redirect?page=changes` => `${{env.BUILD_URL}}changes` AFAICT.

I don't have any strong opinions though.  Easy to close as #wontfix if we like the current setup.

CC @BetsyMcPhail @jwnimmer-tri

jwnimmer-tri, 2023-05-03:
I'm not a buildcop so the people who are should weigh in, but for my money the only thing email is good for is as a timely reminder to go open up https://drake-jenkins.csail.mit.edu/view/Production/ and see what's what.  That page is the way any the light.

svenevs, 2023-05-03:
CC buildcops for low priority bikeshedding conversation: @rpoyner-tri @BetsyMcPhail @liangfok @EricCousineau-TRI @SeanCurtis-TRI @ggould-tri @mwoehlke-kitware @DamrongGuoy @xuchenhan-tri

DamrongGuoy, 2023-05-04:
Do I get it correctly?  The two candidates are:

### A. Series of changes
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/329/display/redirect?page=changes
[image]

### B. Specific to one change
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-continuous-debug/329/changes
[image]

### My preference
I like **B. Specific to one change**, so I don't have to remember the change number (329 in this case).

svenevs, 2023-05-04:
Yes, thank you for clarifying -- pictures were much more valuable here...  Currently we have A, my preference is also B.

SeanCurtis-TRI, 2023-05-04:
I have a small bias for having the context. I rely on that as much as the specific failed build. So, whether I start with context and need to remember which build was specifically flagged (if it isn't obvious by being the only angry build within a recent window) or whether I'm looking at an unambiguously called-out build but have to go fishing for the context, there's the potential for a two-step operation for me. The current seems to carry less friction for how I think about it.

That said, if the majority prefer a more targeted link, I wouldn't grouse.

svenevs, 2023-05-05:
All good points.  It's not particularly obvious how to get to link A on your own and it can be helpful as stated.

When I am expecting a ripple through failure that I know is already fixed, I want link B to be able to quickly confirm "is this the commit I know is already fixed, or do I need to worry more."

Easy solution: I'll just add the link B after link A.  We can have two bicycle  wheels.

BetsyMcPhail, 2023-12-22:
@mwoehlke-kitware while you're editing Jenkins, this issue should be quick to implement.

svenevs, 2024-01-03:
The emails will now include both A and B (`See <${{env.BUILD_URL}}display/redirect?page=changes> and <${{env.BUILD_URL}}changes>`)

closed 2024-01-03 by svenevs