RobotLocomotion/drake #19341: CI: Cache all continuous jobs

BetsyMcPhail, 2023-05-03:
There is currently one continuous job that is not cached: [linux-focal-clang-bazel-continuous-debug](https://drake-jenkins.csail.mit.edu/view/Production/job/linux-focal-clang-bazel-continuous-debug/). This is by design, but the design should be improved. 

In this issue:

- [ ] Simplify the [script](https://github.com/RobotLocomotion/drake-ci/blob/main/driver/environment.cmake) that determines which jobs are cached 
- [ ] Ensure all continuous jobs are cached
- [ ] Delete some continuous jobs (if needed)

Related to #17806 and #19149

For reference, [slack conversation](https://drakedevelopers.slack.com/archives/C270MN28G/p1683132844682249).

jwnimmer-tri, 2023-05-18:
See https://github.com/RobotLocomotion/drake/issues/19421#issue-1714581800 for additional details.

BetsyMcPhail, 2023-05-18:
Current state of caching for Continuous jobs

No caching:

- [linux-focal-clang-bazel-continuous-debug](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-debug/)
- [linux-focal-clang-bazel-continuous-everything-address-sanitizer](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-everything-address-sanitizer/)
- [linux-focal-continuous-cache-server-health-check](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-continuous-cache-server-health-check/)
- [linux-focal-unprovisioned-gcc-bazel-continuous-mirror-to-s3](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-unprovisioned-gcc-bazel-continuous-mirror-to-s3/)
- [linux-focal-unprovisioned-gcc-bazel-continuous-packaging](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-unprovisioned-gcc-bazel-continuous-packaging/)
- [linux-focal-unprovisioned-gcc-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-unprovisioned-gcc-bazel-continuous-release/)
- [linux-jammy-unprovisioned-gcc-bazel-continuous-documentation](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-unprovisioned-gcc-bazel-continuous-documentation/)
- [mac-x86-monterey-clang-bazel-continuous-release](mac-x86-monterey-clang-bazel-continuous-release)
- [mac-arm-monterey-continuous-cache-server-health-check](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-continuous-cache-server-health-check/)

Upload only  (falls under check for DEBUG in bazel.cmake):

- [linux-focal-clang-bazel-continuous-leak-sanitizer](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-leak-sanitizer/)
- [linux-jammy-gcc-bazel-continuous-debug](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-debug/)

Cache (upload & download):
- [linux-focal-clang-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-release/)
- [linux-focal-clang-bazel-continuous-everything-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-everything-release/)
- [linux-focal-gcc-bazel-continuous-everything-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-bazel-continuous-everything-release/)
- [linux-focal-gcc-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-bazel-continuous-release/)
- [linux-focal-gcc-cmake-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-gcc-cmake-continuous-release/)
- [linux-jammy-gcc-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-release/)
- [mac-arm-monterey-clang-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/)
- [mac-arm-ventura-clang-bazel-continuous-non-production-release](https://drake-jenkins.csail.mit.edu/view/Continuous/job/mac-arm-ventura-clang-bazel-continuous-non-production-release/)

BetsyMcPhail, 2023-05-18:
[environment.cmake](https://github.com/RobotLocomotion/drake-ci/blob/main/driver/environment.cmake) needs to be updated with the following logic:

Cache all `release` jobs with the following exceptions:

- Unprovisioned jobs are never cached
- Weekly jobs are never cached
- Mac x86 jobs are never cached

_All_ Pre-merge ([pull request](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/)) jobs should all be cached. non-release pre-merge jobs that should be cached:

- [linux-focal-clang-bazel-experimental-leak-sanitizer](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-focal-clang-bazel-experimental-leak-sanitizer/)
- [linux-jammy-gcc-bazel-experimental-debug](https://drake-jenkins.csail.mit.edu/view/Pull%20Request/job/linux-jammy-gcc-bazel-experimental-debug/)

non-release continuous jobs that should be cached:

- [linux-focal-clang-bazel-continuous-leak-sanitizer](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-leak-sanitizer/)
- [linux-focal-clang-bazel-continuous-debug](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-debug/)
- [linux-jammy-gcc-bazel-continuous-debug](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-debug/)

non-release nightly jobs that should be cached:

- [linux-focal-clang-bazel-nightly-leak-sanitizer](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-focal-clang-bazel-nightly-leak-sanitizer/)
- Note: The nightly equivalent of the cached `continuous-debug` jobs don't exist

[cmake.cmake](https://github.com/RobotLocomotion/drake-ci/blob/main/driver/configurations/cmake.cmake) and [bazel.cmake](https://github.com/RobotLocomotion/drake-ci/blob/main/driver/configurations/bazel.cmake) need to define that:
- experimental cached jobs read from the cache
- continuous cached jobs read from and write to the cache
- nightly cached jobs write to the cache

jwnimmer-tri, 2023-05-19:
BTW I think the write-up would be clearer if is was phrased as two separate choices:
- Which jobs write to the cache.
- Which jobs read from the cache.

To me, trying to explain both at once leads to confusion.  I suspect that the CMake code to govern the on/off would also benefit from reading read & write as distinct.

jwnimmer-tri, 2023-05-19:
It would probably also help if the CMakeLists had a single source of truth for "Should this job read? Should this job write?" and then *separately* the bazel + cmake listfiles would map that decision into how to spell it on the command line.

\CC @mwoehlke-kitware FYI

mwoehlke-kitware, 2023-05-19:
Jobs look like `<platform-and-compiler>-<track>-<type>`. IIUC that whether to use the cache in some fashion is strictly dependent on platform/compiler/type without regard to track, then I think it's reasonable to keep the "uses cache" and "uses cache *how*" decisions separate. However, there should be a comment to that effect at the start of the "do I use the cache?" logic. (I think it might also be good to change the regular expressions to use `${ANY_TRACK}` rather than repeating `(nightly|continuous|experimental)`.)

Also, I think we should be careful about exactly matching the logic to the set of existing jobs. For example, if `linux-focal-clang-bazel-nightly-debug`, would we want it cached? If not, that should probably be clearly stated. Otherwise, I think it would be clearer if the logic is written so that that's a cached build, even though it doesn't exist. The reader shouldn't have to know that job doesn't exist to not wonder why it isn't cached.

jwnimmer-tri, 2023-05-19:
Good points.  I think maybe the happy place is partway in the middle of our ideas, if I've understood correctly.

(1) I think the predicate we want for "should read from cache" is the jobs that trigger automatically in pre-merge or continuous, excluding unsupported archs (mac x86) and excluding unwanted flavors (unprovisioned).

(2) I think the predicate we want for "should write to cache" is any jobs triggered in continuous or nightly that are congruent to some "should read from cache" build in all aspects other than the triggering.

The overall goal is to ensure that all pre-merge and post-merge builds are as efficient as possible, by having them read from the cache (item 1), and having something usefully populate that cache (item 2).

We also don't want to waste time populating a cache that will never be read from, which is why (2) is conditioned on jobs that (1) actually reads from.

mwoehlke-kitware, 2023-05-19:
Okay, I think we should probably keep the current 'determinations based on platform/compiler/type first, *but* add a comment explaining we're only deciding whether to use the cache *in some capacity*, and that whether to read, write or both is done <reference to elsewhere>. IMHO, we don't need to move 'elsewhere' (though I'm not opposed to doing so, either) so long as that comment exists. Possibly that comment could include Jeremy's explanation of how to decide what jobs should use cache.

BetsyMcPhail, 2023-05-19:
The current set of jobs that read from the cache is larger than just a subset of pre-merge and continuous. It also includes all jobs that end in "-release" with the exclusions mentioned above . For example, https://drake-jenkins.csail.mit.edu/view/Experimental/job/linux-focal-gcc-cmake-experimental-everything-release/. 

There is also a special case for https://drake-jenkins.csail.mit.edu/view/Mac%20ARM/job/mac-arm-monterey-clang-bazel-experimental-debug/.

Is this by design? An unintended side effect? Either way, is it valuable? I'm not sure.

jwnimmer-tri, 2023-05-19:
Ah, interesting.  A job that isn't exactly one of the pre-merge + continuous wouldn't expect 100% cache hits, but neither would it expect 0% -- some build actions will be the same across related configurations.  So there might still be value in having those read from the cache, so that hand-triggered PR builds would be faster than otherwise.  I'll need to think on this a bit more.

BetsyMcPhail, 2023-05-19:
To make it completely clear, the nightly versions of the "extra" experimental jobs also write to the cache.

jwnimmer-tri, 2023-05-19:
Yup.  I'll take some time to think this all over, but my first reaction is that those are a waste of cache space.  The cache should be tuned to optimize the common case.  Off-nominal builds are rare enough that we don't need them to be speedy.

jwnimmer-tri, 2023-05-19:
It might also be worth opening the question of why Nightly doesn't read from the cache.  I imagine it was for "safety of a clean build" reasons, but I'm sure the hazard of a lying cache is so large as to be worth worrying about.  We have Unprovisioned for a truly clean build from scratch.  Still thinking.

jwnimmer-tri, 2023-06-22:
Here's my current thinking:

(1) All jobs should read from the cache, unless there is an important reason not to.  Specifically, the jobs that must **not** read from the cache are:
- (a) unsupported archs (i.e., mac x86),
- (b) non-build jobs (i.e., health check)
- (b) clean builds (i.e., unprovisioned),
- (c) no-sandbox builds that are probably incompatible with caching (i.e., coverage).

Everything else is safe to read from the cache, so we might as well turn it on -- the trigger type doesn't matter.  Trying to fine-tune it beyond this broad stroke seems like too much config work.  Even if there are practically no cache hits in some config, I don't expect too much pain from having it enabled.

(2) All continuous jobs that read from the cache should also write to the cache.  We want Continuous to be fast.

(3) Notably, the pull request builds are a subset of Continuous.  Therefore, they should always have a warm cache available (from the Continuous jobs).

(4) Notably, nightly should not write to the cache.

---

I don't mind however you prefer to encode that into the CI scripts.

closed 2023-07-05 by svenevs