RobotLocomotion/drake #19346: ModuleNotFoundError: No module named 'pydrake.planning.common_robotics_utilities'

dbdxnuliba, 2023-05-04:
### What happened?

>>> from IPython.display import HTML, SVG
>>> from pydrake.geometry import MeshcatVisualizer, MeshcatVisualizerParams, Role, StartMeshcat
>>> from pydrake.geometry.optimization import IrisInConfigurationSpace, IrisOptions
>>> from pydrake.multibody.parsing import LoadModelDirectives, Parser, ProcessModelDirectives
>>> from pydrake.multibody.plant import AddMultibodyPlantSceneGraph
>>> from pydrake.planning.common_robotics_utilities import SimpleRRTPlannerState
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pydrake.planning.common_robotics_utilities'
>>> import pydrake.planning
>>> 

### Version

drake-1.13.0

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using CMake

### Relevant log output

```shell
cd /home/dell/ref/drake-1.13.0
mkdir drake-build
cd drake-build
cmake -DWITH_GUROBI=OFF -DWITH_MOSEK=ON -DWITH_SNOPT=ON ..
make -j2

and finish install it into /home/dell/ref/drake-1.13.0/drake-build/install

and then sudo gedit ~/.bashrc
#bazel related
source /home/dell/.bazel/bin/bazel-complete.bash
export PATH=$PATH:$HOME/bin

export DRAKE_BUILD_DIR=$HOME/ref/drake-1.13.0/drake-build
export PYTHONPATH=${DRAKE_BUILD_DIR}/install/lib/python3.8/site-packages:${PYTHONPATH}
```

jwnimmer-tri, 2023-05-04:
>  ModuleNotFoundError: No module named 'pydrake.planning.common_robotics_utilities'

There has never been such a module as part of Drake.  I am not sure where you found that line, but maybe it was from someone who was using a fork of Drake.

Do you have any more details you could offer?

dbdxnuliba, 2023-05-05:
@jwnimmer-tri 
I find here https://github.com/mpetersen94/gcs/blob/main/reproduction/prm_comparison/prm_comparison.ipynb
[image]

jwnimmer-tri, 2023-05-05:
Okay, thanks for the link.  That example from `mpetersen94` repository must not have been using an official version of Drake.  Perhaps you can ask over in that repository how to run that reproduction.

In any case, there is no bug in Drake.  There has never been `pydrake.planning.common_robotics_utilities` module as part of Drake.

closed 2023-05-05 by jwnimmer-tri

dbdxnuliba, 2023-05-06:
OK ,thanks a lot