RobotLocomotion/drake #19348: CI packaging build clarity

jwnimmer-tri, 2023-05-04:
See example from https://github.com/RobotLocomotion/drake/pull/19343#issuecomment-1533610747.

I launched packaging builds, and manually summarized their results as a "how to run a test install" for users.  That involved reading through the build log per the https://drake.mit.edu/jenkins.html#building-packages-on-demand instructions and then copy-pasting a bunch of stuff.

One easy improvement would be at the _very end_ of the CI build log on a packaging job, we print the URLs all in once place (instead of staggered throughout the log).  Even better, print the _install instructions_ in full (i.e., `pip install https://...` or `curl -fsSLO https://...` etc).

Even better (thought possibly more worn than its worth), if _Jenkins_ could post a PR comment with that information when the build completes, that would be the most convenient.

mwoehlke-kitware, 2023-10-03:
Jenkins posting a comment would only apply to experimental builds, yes? For other builds, we'd still need the end-of-log summary, I think? (I agree it's a good idea though!)

jwnimmer-tri, 2023-10-03:
Yes -- in particular, the comment posting would only apply pull requests (not all experimental packaging or wheel jobs -- sometimes we launch experimental jobs that aren't associated with any pull request).

I assume that posting back the comment when the job is done might be fairly difficult.  I'd be satisfied with only the additional log text to resolve this issue.

jwnimmer-tri, 2023-10-12:
In the summary at the end, we should _not_ print the sha256/sha512 links in the shortcut tips.  (Leaving it in the output stream is fine.)

mwoehlke-kitware, 2023-10-23:
Proposed fix: https://github.com/RobotLocomotion/drake-ci/pull/250

closed 2023-10-24 by mwoehlke-kitware