RobotLocomotion/drake #19365: SetPositions(): condition 'q.size() == num_positions()' failed.

dvogureckiy99, 2023-05-05:
### What happened?

When I tried run urdf model with overall joints number grater than actuated joints I have got error: 
```
Failure at bazel-out/k8-opt/bin/tools/install/libdrake/_virtual_includes/drake_shared_library/drake/multibody/plant/multibody_plant.h:2138 in SetPositions(): condition 'q.size() == num_positions()' failed.
```

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using CMake

jwnimmer-tri, 2023-05-05:
This bug report does not contain enough details for anyone to help you.  Please read https://www.chiark.greenend.org.uk/~sgtatham/bugs.html and provide more information along those lines.  At minimum, you need to provide instructions that reproduce the problem, including which version of Drake you're using and how to run the failing program.

jwnimmer-tri, 2023-05-05:
Thanks for editing the post with more information, but it's still not enough:

```
jwnimmer@call-cps:~/tmp/quadruped_drake$ ../env/bin/python3 ./simulate.py 
Traceback (most recent call last):
  File "/home/jwnimmer/tmp/quadruped_drake/./simulate.py", line 103, in <module>
    planner = builder.AddSystem(TowrTrunkPlanner(trunk_frame_ids))
  File "/home/jwnimmer/tmp/quadruped_drake/planners/towr.py", line 29, in __init__
    self.GenerateTrunkTrajectory()
  File "/home/jwnimmer/tmp/quadruped_drake/planners/towr.py", line 60, in GenerateTrunkTrajectory
    sub.Popen(["build/towr/trunk_mpc","walk","0","1.5", "0.0"], env=my_env)  
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'build/towr/trunk_mpc'
```

It looks like maybe I need to run a C++ build as well?

Please provide reproduction instructions that succeed starting from zero (e.g., they would work in a Docker container).

If you can distill the bug down to a single Python program that only uses pydrake and nothing else, that's even better.

dvogureckiy99, 2023-05-06:
> Thanks for editing the post with more information, but it's still not enough:
>
> ```
> jwnimmer@call-cps:~/tmp/quadruped_drake$ ../env/bin/python3 ./simulate.py
> Traceback (most recent call last): [...]

sorry, I impoved my question, I forgot to mention compiling and dependencies.

jwnimmer-tri, 2023-05-06:
>you need install all this dependencies:
>-- [Eigen](http://eigen.tuxfamily.org/)
>-- [Ipopt](https://projects.coin-or.org/Ipopt)
>-- [ifopt](https://github.com/ethz-adrl/ifopt)
>-- [CMake](https://cmake.org/cmake/help/v3.0/)
>-- [Numpy](https://numpy.org/)

This is too much work.

For one thing, "Install these dependencies" is not a specific enough instruction.  The bug report should contain **an exact list of bash commands to run** to avoid any confusion, and to allow us to reproduce the problem without any struggle or ambiguity.

Even with that, asking us to build and install random extra libraries is asking too much effort for the Drake maintainers.  Instead, please provide a reproducible example that depends only pydrake and packages from `pip`, so that the preparation for reproducing the bug is as simple as `pip install A B C D` and then `python3 demoprogram.py`.

---

In the alternative, you could provide a more thorough backtrace so we can start to get a picture of what's happening.

The first question for this bug will be -- which code is calling `SetPositions()` with a wrong-sized `q` vector?  If the call is coming from inside Drake, we'll need to track that down and at least provide a better error message.

If the call to `SetPositions()` is coming from your code, then this isn't a bug in Drake.  You're just passing the wrong number of positions.  In that case we can improve Drake at least report the specific `q.size()` and `plant.num_positions()` to make the error message better, but the root cause fix will be in your own code.

dvogureckiy99, 2023-05-06:
@jwnimmer-tri 
Sorry, `SetPositions()` is coming from my code. It will be very usefull if this error message

jwnimmer-tri, 2023-05-06:
> It will be very useful if this error message report the specific q.size() and plant.num_positions().

That's a great suggestion, I'll re-title the issue to match that.

> I don't know how to actuate only some specific joints, when system is underactuated.

Yes, for questions about using Drake, please check https://stackoverflow.com/questions/tagged/drake.  There are already a few Q&As there related to underactuated robots.  If there's no question already there which matches your situation, please post a new question there with details.

I'll keep this issue open to track the better error message.

dvogureckiy99, 2023-05-06:
@jwnimmer-tri 
I wanna try solve this problem, but how can I speed up the compiling process so that I don't need to compile all the modules, becaudr it takes so long time?

jwnimmer-tri, 2023-05-06:
Did you see https://drake.mit.edu/installation.html?  There are compiled binaries available in a wide variety of formats.

dvogureckiy99, 2023-05-06:
> 

Yes, I see, but whether I can change this binaries in order to make better error message ? I thought no.

dvogureckiy99, 2023-05-06:
@jwnimmer-tri 
I tried change `PrintFailureDetailTo` function in [my local commit](PrintFailureDetailTo) , but it got error, cause 'q' was not declared in this scope and so I cannot  bring out `q` value.

jwnimmer-tri, 2023-05-06:
To debug your own code `quadruped_drake`, there is no need to rebuild or change Drake.  You can use the precompiled binaries.  At the point where `quadruped_drake` code is calling `SetPositions`, you can yourself print out `plant.num_positions()` and `len(q)`, or check if they match and throw a nice error, or etc.

For improving Drake to produce a better error message in the future, we're going to wait for #19229 to finish and then we can use that new feature the MultibodyPlant.

ggould-tri, 2023-05-08:
Assigning @amcastro-tri as component owner (assuming you don't want to grab this, @jwnimmer-tri?).