RobotLocomotion/drake #19394: //bindings/pydrake/systems:py/lcm_test failed in mac CI

ggould-tri, 2023-05-11:
See https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-monterey-clang-bazel-nightly-everything-release/216/console

```
//bindings/pydrake/systems:py/lcm_test (see /Users/admin/workspace/mac-arm-monterey-clang-bazel-nightly-everything-release/_bazel_admin/742ea99c3dc23bbb0252a6111ea82360/execroot/drake/bazel-out/darwin_arm64-opt/testlogs/bindings/pydrake/systems/py/lcm_test/test.log)
[4:52:26 AM]  ==================== Test output for //bindings/pydrake/systems:py/lcm_test:
[4:52:26 AM]  
[4:52:26 AM]  Test bindings of LCM integration with the Systems framework.
[4:52:26 AM]  
[4:52:26 AM]  
[4:52:26 AM]  
[4:52:26 AM]  Running tests...
[4:52:26 AM]  ----------------------------------------------------------------------
[4:52:26 AM]  .INFO:drake:LCM bus 'bar' created for URL memq://2
[4:52:26 AM]  INFO:drake:LCM bus 'foo' created for URL memq://1
[4:52:26 AM]  ............E
[4:52:26 AM]  ======================================================================
[4:52:26 AM]  ERROR [0.031s]: test_subscriber_wait_for_message (lcm_test.TestSystemsLcm.test_subscriber_wait_for_message)
[4:52:26 AM]  Checks how `WaitForMessage` works without Python threads.
[4:52:26 AM]  ----------------------------------------------------------------------
[4:52:26 AM]  Traceback (most recent call last):
[4:52:26 AM]    File "/Users/admin/workspace/mac-arm-monterey-clang-bazel-nightly-everything-release/_bazel_admin/742ea99c3dc23bbb0252a6111ea82360/sandbox/darwin-sandbox/15874/execroot/drake/bazel-out/darwin_arm64-opt/bin/bindings/pydrake/systems/py/lcm_test.runfiles/drake/bindings/pydrake/systems/test/lcm_test.py", line 194, in test_subscriber_waiINFO: From Testing //bindings/pydrake/systems:py/lcm_test:
[4:52:26 AM]  t_for_message
[4:52:26 AM]      new_count = sub.WaitForMessage(
[4:52:26 AM]                  ^^^^^^^^^^^^^^^^^^^
[4:52:26 AM]    File "/Users/admin/workspace/mac-arm-monterey-clang-bazel-nightly-everything-release/_bazel_admin/742ea99c3dc23bbb0252a6111ea82360/sandbox/darwin-sandbox/15874/execroot/drake/bazel-out/darwin_arm64-opt/bin/bindings/pydrake/systems/py/lcm_test.runfiles/drake/bindings/pydrake/systems/_lcm_extra.py", line 31, in Deserialize
[4:52:26 AM]      message = self._lcm_type.decode(buffer)
[4:52:26 AM]                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[4:52:26 AM]    File "/Users/admin/workspace/mac-arm-monterey-clang-bazel-nightly-everything-release/_bazel_admin/742ea99c3dc23bbb0252a6111ea82360/sandbox/darwin-sandbox/15874/execroot/drake/bazel-out/darwin_arm64-opt/bin/bindings/pydrake/systems/py/lcm_test.runfiles/drake/lcmtypes/drake/lcmt_quaternion.py", line 40, in decode
[4:52:26 AM]      raise ValueError("Decode error")
[4:52:26 AM]  ValueError: Decode error
[4:52:26 AM]  
[4:52:26 AM]  ----------------------------------------------------------------------
[4:52:26 AM]  Ran 14 tests in 0.049s
[4:52:26 AM]  
[4:52:26 AM]  FAILED (errors=1)
```

This has only happened once, so closing for now.

DamrongGuoy, 2023-05-11:
I'll set [“component: continuous integration” and “type: bug”](https://github.com/RobotLocomotion/drake/issues?q=is%3Aissue+is%3Aopen+label%3A%22component%3A+continuous+integration%22+label%3A%22type%3A+bug%22) according to [Drake Build Cop Review and Retrospective](https://docs.google.com/document/d/120AOAaamIMO-SM1UaJ6vfzpA15LnXHexDF4a7MLAS3o/edit#heading=h.u4sg5dei1fn3).

DamrongGuoy, 2023-05-11:
Sorry.  I misread.  It's not recurring.  Remove the labels.
```
Recurring build failures should be tracked as issues with “component: continuous integration” 
and “type: bug” which should be the authoritative source of build failure information.
```

DamrongGuoy, 2023-05-19:
It failed again back to back on May 18, 2023:

- [mac-arm-monterey-clang-bazel-continuous-release/721/](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/721/)
- [mac-arm-monterey-clang-bazel-continuous-release/722/](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/722/)

Repeating the run again passed:
- [mac-arm-monterey-clang-bazel-continuous-release/723/](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/723/)

Set "component: continuous integration” and “type: bug”. Reopen.

DamrongGuoy, 2023-05-19:
Failed again on May 19, 2023:

- [mac-arm-monterey-clang-bazel-continuous-release/730/](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/730/)

Repeating it passed: 
- [mac-arm-monterey-clang-bazel-continuous-release/731/](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/731/)

BetsyMcPhail, 2023-05-25:
May 25th failure on Ventura: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-everything-release/3/

svenevs, 2023-06-02:
Another instance on [2023-06-02](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-monterey-clang-bazel-nightly-everything-release/240/consoleFull).

BetsyMcPhail, 2023-06-02:
June 1: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-ventura-clang-bazel-continuous-release/35/

svenevs, 2023-06-05:
- June 4: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-monterey-clang-bazel-nightly-everything-release/243/consoleFull
- June 5: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-monterey-clang-bazel-nightly-everything-release/244/consoleFull

jwnimmer-tri, 2023-06-05:
Oops.  The flaky test was `//bindings/pydrake/systems:py/lcm_test` not `//bindings/pydrake:py/lcm_test`.  The PR #19531 marked the wrong test.

BetsyMcPhail, 2023-06-06:
Oh no, I'll create a fixup PR

PR #19541 (2023-06-06): [pydrake] Mark systems lcm_test as flaky
Fix error in earlier commit where the wrong test was marked as flaky.

Resolves #19394

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19541)
closed 2023-06-06 by jwnimmer-tri