RobotLocomotion/drake #19421: Enable remote cache downloads in Continuous Production

jwnimmer-tri, 2023-05-17:
See [here](https://github.com/RobotLocomotion/drake-ci/blob/f4f81402841840ca918a0d2079f3b31e866a183c/driver/configurations/bazel.cmake#L112-L113) for:

```cmake
  elseif(DASHBOARD_TRACK STREQUAL "Continuous" AND DEBUG)
    set(DASHBOARD_REMOTE_ACCEPT_CACHED "no")
```

This seems to me to be vestigial from https://github.com/RobotLocomotion/drake-ci/commit/80b53e4853c90c8b2dc1f55d93a8580f0979b02f when bazel and our cache server were bad at their jobs.  I don't think it's likely to be a problem anymore.

I'd like to try removing this stanza (i.e., to enable caching on https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-jammy-gcc-bazel-continuous-debug and https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-focal-clang-bazel-continuous-debug), and see if it falls over.

BetsyMcPhail, 2023-05-18:
This seems like a duplicate of #19341? Although that issue is not quite accurate, it only mentions `linux-focal-clang-bazel-continuous-debug`. With some further digging, linux gcc and clang debug jobs have different caching rules: https://github.com/RobotLocomotion/drake-ci/blob/f4f81402841840ca918a0d2079f3b31e866a183c/driver/environment.cmake#L159

jwnimmer-tri, 2023-05-18:
> This seems like a duplicate of https://github.com/RobotLocomotion/drake/issues/19341?

Right!

> ... gcc and clang debug jobs have different caching rules ...

Note that `environment.cmake` governs caching for CMake jobs, and `bazel.cmake` governs caching for Bazel jobs.

closed 2023-05-18 by jwnimmer-tri