RobotLocomotion/drake #19471: Why Meldis visualizer is freezing during simulation?

dvogureckiy99, 2023-05-23:
### What happened?

I trying run `deformablesdrake/deformable_torus.ipynb` project, but when I am running `python3 -m pydrake.visualization.meldis` Meldis visualizer it's freezing during simulation, during `simulator.AdvanceTo(simulation_time)`.

### Version

the last source from git

### What operating system are you using?

Ubuntu 20.04

### What installation option are you using?

compiled from source code using CMake

xuchenhan-tri, 2023-05-23:
There is no `deformable_torus.ipynb` in Drake but I assume the thing you are running is the python script provided in the issue.

I was able to run the script you provided locally (not in a notebook) just fine. Meldis is able to visualize a deformable torus dropped from height and lands on a rigid body weld to the world and jiggles.

Could you describe in more details of what you mean by "freezing"? The `while 1` loop looks a bit suspicious to me and it might not do what you intend to do. To be clear, this doesn't put the simulation in a loop. The expected effect is that the simulation runs to time `simulation_time` and stops there.

dvogureckiy99, 2023-05-24:
@xuchenhan-tri 
1. I converted this script from `.ipynb` to `.py`
2. Deleted `while 1` loop.
3. and runned it locally, but the problem remains the same: 
during the simulation, my cursor freezes and any commands from the periphery are not processed, after a couple of seconds everything is defrosted, but at the same time the simulation continues not from the moment of hanging, but after a few seconds. 
If it were possible to loop the simulation in Meldis, then this problem would be solved, since freezes after a few seconds pass.

xuchenhan-tri, 2023-05-24:
> In Meshcat I solved this problem by looping simulation

Is it true that the freezing also happens when you run Meshcat with a rigid body simulation?

This sounds to me like a Meshcat issue more than a deformable body simulation issue.

dvogureckiy99, 2023-05-25:
@xuchenhan-tri 

Yes, it is freezing even with the rigid bodies.

jwnimmer-tri, 2023-05-25:
Okay, that sounds like a problem particular to your computer.  We do not see any freezing when we run these things.

For us to help with fixing it, ideally we would be able to reproduce the problem.  Is there anything you can think of about your computer that might be unique, that we could try to replicate on our side?  A particular web browser product or version?  Running in a virtual machine?  Discrete graphics card versus built-in Intel graphics?

Also, please try running in a private window (so that all browser extensions and add-ons are disabled).

dvogureckiy99, 2023-05-25:
@jwnimmer-tri 
Thank you.
I am using:
Mozilla Firefox for ubuntu 113.0.2 (64-bit)
Running on nonvirtual machine Ubuntu 20.04
Discrete graphic card NVIDIA GeForce 920MX

 I tried run in a private window, but freezing the same as before.

jwnimmer-tri, 2023-05-25:
Hm.  We have similar things here with no trouble.

We are usually Ubuntu 22 (not 20), but it is unlikely that Ubuntu version is related to the problem though so I'm not worried about that part.

I suggest:

(1) Make sure the nvidia drivers are installed correctly from Ubuntu, e.g., something like one of these is installed:
https://packages.ubuntu.com/focal/nvidia-driver-525
or
https://packages.ubuntu.com/focal/nvidia-driver-530

We don't typically run using https://packages.ubuntu.com/jammy/xserver-xorg-video-nouveau so if you are using Nouveau that might be related to the problem.

(2) Make sure that you have rebooted in case the driver has been upgraded recently (even in the background by auto-update).

dvogureckiy99, 2023-05-25:
@jwnimmer-tri  yes, I am using Nouveau, but how to solve this ?

dvogureckiy99, 2023-05-26:
I tried today run this file on GPU GeForce RTX 3080 and it turns out simulating without freezing.

jwnimmer-tri, 2023-06-07:
I don't think there's much we can do to debug computer-specific problems.

We are open to pull requests if anyone whose computer is struggling can figure out the root cause and provide a solution.

closed 2023-06-07 by jwnimmer-tri