RobotLocomotion/drake #19479: Document recommendations for converting mesh assets into drake-compatible formats

RussTedrake, 2023-05-24:
- DAE => OBJ is discussed in #19109. 
- @DamrongGuoy has a working document for preparing non-convex meshes for hydroelastic.

Per the [discussion on slack](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1683934761753059?thread_ts=1683917072.132939&cid=C2WBPQDB7), @DamrongGuoy has agreed to take this one on.

DamrongGuoy, 2023-05-24:
@jbinagia @JShep-tri and I are working on this public doc:

- https://docs.google.com/document/d/1VZtVsxIjOLKvgQ8SNSrF6PtWuPW5z9PP7-dQuxfmqpc/edit?usp=sharing
 
I hope it will go into Drake's Doxygen later.

DamrongGuoy, 2023-05-24:
Relevant https://github.com/RobotLocomotion/models/issues/24

jwnimmer-tri, 2024-07-09:
This is a weird issue.  The OP lists two topics:

(1) Tips for DAE, which is already comprehensively tracked in a different issue).

(2) Document for preparing non-convex meshes for hydroelastic.

Since the two topics are basically unrelated, and we already have a tracking issue for the first, wouldn't it make the most sense to reformulate this issue as something like "Provide instructions for how to create tetrahedral volume meshes as `*.vtk` files"?

closed 2024-11-02 by jwnimmer-tri