RobotLocomotion/drake #19483: DirectCollocation constructor arguments TypeError

matheecs, 2023-05-24:
### What happened?

Failed to call `DirectCollocation(plant, plant_context, ...)` in python

```python
from pydrake.examples import PendulumPlant
from pydrake.planning import DirectCollocation
from pydrake.systems.framework import InputPortSelection

plant = PendulumPlant()
context = plant.CreateDefaultContext()

num_time_samples = 21
dircol = DirectCollocation(
    plant,
    context,
    num_time_samples=num_time_samples,
    minimum_time_step=0.2,
    maximum_time_step=0.5,
    input_port_index=InputPortSelection.kUseFirstInputIfItExists,
    assume_non_continuous_states_are_fixed=False)
prog = dircol.prog()
num_initial_vars = prog.num_vars()
```

Result

```text
TypeError                                 Traceback (most recent call last)
Cell In[1], line 10
      6 context = plant.CreateDefaultContext()
      9 num_time_samples = 21
---> 10 dircol = DirectCollocation(
     11     plant,
     12     context,
     13     num_time_samples=num_time_samples,
     14     minimum_time_step=0.2,
     15     maximum_time_step=0.5,
     16     input_port_index=InputPortSelection.kUseFirstInputIfItExists,
     17     assume_non_continuous_states_are_fixed=False)
     18 prog = dircol.prog()
     19 num_initial_vars = prog.num_vars()

TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pydrake.planning.DirectCollocation(system: pydrake.systems.framework.System, context: pydrake.systems.framework.Context, num_time_samples: int, minimum_timestep: float, maximum_timestep: float, input_port_index: Union[pydrake.systems.framework.InputPortSelection, pydrake.systems.framework.InputPortIndex] = , assume_non_continuous_states_are_fixed: bool = False, prog: pydrake.solvers.MathematicalProgram = None)

Invoked with: , ; kwargs: num_time_samples=21, minimum_time_step=0.2, maximum_time_step=0.5, input_port_index=, assume_non_continuous_states_are_fixed=False
```

### Version

v1.17.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

jwnimmer-tri, 2023-05-24:
Note the error message:

> The following argument types are supported: ... minimum_timestep: float ...
> Invoked with: ... minimum_time_step=0.2 ...

See `time_step` vs `timestep`.

In #19470 that function argument was renamed.

The website API reference https://drake.mit.edu/pydrake/index.html reflects the nightly build (`pip install https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-cp38-cp38-manylinux_2_31_x86_64.whl`), not the stable release (`pip install drake`).

jwnimmer-tri, 2023-05-24:
Closing as duplicate of the (just now created) feature request to publish the stable release docs on the website, #19484.

closed 2023-05-24 by jwnimmer-tri