RobotLocomotion/drake #19485: Bug with Meshcat Slider Bounds and Step Size

cohnt, 2023-05-24:
### What happened?

If I want to construct a meshcat slider that has a minimum value of -1, a maximum value of 1, and a step size of 2, such that the only values it can take are -1 and 1, I should be able to do so with `meshcat.AddSlider("foo", value=1, min=-1, max=1, step=2)`. However, this leads to the slider's initial value being 2, and dragging it to the left sets it to 0. I also can't manually specify the value to -1 or 1, as it's immediately truncated to 0 and 2 (respectively).

jwnimmer-tri, 2023-05-24:
I can confirm I see the same thing.

I logged the values that our `meshcat.cc` is transmitting, and they are correct (i.e., no loss of precision or roundoff errors).  So this is a bug in either https://github.com/rdeits/meshcat or the widget toolkit that it's using.

We can keep the Drake bug open for tracking, but the fix pull request will be somewhere upstream.

jwnimmer-tri, 2023-05-24:
It's probably obvious, but the work-around is to use bounds without the bug (min=0, max=1, step=1) and then convert at the call site `2 * meshcat.GetSliderValue("foo") - 1`.

cohnt, 2023-05-24:
Yes, that's exactly the workaround I've been using.

cohnt, 2023-07-21:
Any updates on this issue? Should I move discussion to the meshcat repo?

jwnimmer-tri, 2023-07-21:
No updates from the TRI side; this isn't on our near-term roadmap to fix yet.

Feel free to open an issue on the Meshcat repo, but I'm not sure that anyone beyond TRI is listening there, either.

If you develop a fix, we do have TRI bandwidth to help review and merge it.

closed 2024-11-01 by jwnimmer-tri