RobotLocomotion/drake #19494: Publishing contact visualization to MeshCat can slow down gamepad response

rcory, 2023-05-25:
### What happened?

If you have an interactive simulation that uses MeshCat's gamepad interface, turning on contact visualization can significantly slow down MeshCat's gamepad response. Anecdotally, the issue seems to worsen as the number of hydrelastic contacts goes up and the visualization publish period decreases (higher frame rates). Turning off contact publishing altogether seems to resolve the issue.

Running [this](https://github.shared-services.aws.tri.global/rick-cory/clutter-drake) demo, and increasing the visualization publish period [here](https://github.shared-services.aws.tri.global/rick-cory/clutter-drake/blob/6a9724c2782ca2938afa9c4d89a925621595062a/clutter_teleop_sim.py#L203) reproduces the behavior.

### Version

1.16.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

jwnimmer-tri, 2023-05-25:
Bug reports on Drake should have public reproducers.  Linking to private repositories is fine to bootstrap the bug report but someone will need to make a public repro for this to remain an open and useful ticket.

rcory, 2023-05-25:
Sure, I can work on that.

DamrongGuoy, 2023-05-25:
Does anyone know how to quantitatively measure MeshCat's gamepad response?

jwnimmer-tri, 2023-05-25:
I assume you want to do this while also running some other simulation code?

I suggest adding a non-scene graph geometry source that postures the 2d angular rotation of a vertical cylinder to display the reported deflection of the gamepad.  In other words, draw the gamepad stick on the screen, using the axes returned by GetGamepad.  Basically connecting up GetGamepad output to the SceneGraph kinematics input for your custom source.

Then when you deflect the stick, you can see how quickly the browser scene changes.  I *think* that's probably the round-trip facet that you care about.

DamrongGuoy, 2023-05-31:
Update.  We (@rcory @xuchenhan-tri @DamrongGuoy) are trying to remove MultibodyPlant from the equation.  I am looking at https://github.com/RobotLocomotion/drake/tree/master/geometry/profiling#contact_surface_rigid_bowl_soft_ballcc. It sends SceneGraph's contact surfaces to LCM without MultibodyPlant.

For Yak shaving, I found this issue
- #19504.

I proposed this 7-line PR
- #19505

Now that we have a Drake example to work with, we'll add Gamepad to this non-MultibodyPlant as the next step.

jwnimmer-tri, 2023-05-31:
Almost certainly the problem is the message queue filling up, I would have thought?  Somewhere between `meshcat.cc` and the browser javascript processing there is more data happening than it can handle.

DamrongGuoy, 2023-05-31:
Thank you @jwnimmer-tri .  Is there a simple indicator for that?  For example, use `top` command and look for memory or network usage of some processes?

I, too, had a hunch that it's about a large number of contacts times a large number of polygons.

jwnimmer-tri, 2023-05-31:
>  Is there a simple indicator for that?

Best guess: no, it would probably involve debugging the software itself.  It's _possible_ that netstat would show the kernel queues backing up, but I would guess probably not for a localhost connection.  Might be worth a look though.

In the slack exposition of this, I understood the slowdown happened only with really aggressive publish rates like full contact patches reloading every frame at 60+ Hz?  Does this ticket deserve a lot of attention?  On first flush, the simple answer is "don't do that".  Why don't we turn down contact patches to 32 Hz by default?  Problem solved?

Another thing to try -- don't poll for gamepad so often.  IIRC, polling for the gamepad blocks a fairly central mutex.  If you are polling too fast, you might be starving the transmit queue, too.  Try polling for the gamepad at 10 Hz or less.

rcory, 2023-05-31:
> In the slack exposition of this, I understood the slowdown happened only with really aggressive publish rates like full contact patches reloading every frame at 60+ Hz? Does this ticket deserve a lot of attention?

@jwnimmer-tri you might have missed my last Slack [update on this](https://tri-internal.slack.com/archives/C2BMF3K28/p1685464957829129?thread_ts=1684987173.698529&cid=C2BMF3K28). The issue seems to be present for our Stanford users running the visualization publishing at 20Hz.

> Another thing to try -- don't poll for gamepad so often

Should be easy to try this. ATM we are polling the gamepad at 20Hz, but could probably crank this way down.

rcory, 2023-05-31:
The clutter sim is now available in [this](https://github.com/rcory/drake/tree/clutter_sim) Drake branch. To run it do `bazel run //examples/clutter:clutter_demo`

DamrongGuoy, 2023-06-01:
Thank you for providing an example in Drake. It helps a lot.

The current hypothesis is that the simulation publishes contact surfaces so much faster than MeshCat can handle.

Today @rcory and I made progress on one of two complaints: _"delay grows as the sim continues"._ 
After this [line](https://github.com/rcory/drake/blob/ede91dcd420320bd8e1e99b2b5de1c70d01c4d3c/examples/clutter/clutter_teleop_sim.py#L126),  add:
```C++
self._meshcat.Flush()
```
It blocks the simulation and gives MeshCat a chance to catch up, preventing the delay to grow.

We still haven't solved the main complaint: _"response time from gamepad input to change in Meshcat pose."_
The `self._meshcat.Flush()` only prevents, for example, a 3-second response time from growing to 30 seconds; however, the 3 seconds itself is too long. We're still working on that one.

DamrongGuoy, 2023-06-01:
Still need confirmation: running MeshCat on another computer (URL = IP address/7000) in the same local network seems to help. My hypothesis is that MeshCat draws contact surfaces slower than network speed. By having a dedicated machine for MeshCat, it doesn't have to share resources with the simulation.

DamrongGuoy, 2023-06-02:
@SeanCurtis-TRI and I did an interesting experiment today (2023-06-01):

- Use a longer publish period of contact visualization than the publish period of geometry poses. It's done by changing [this line](https://github.com/RobotLocomotion/drake/blob/master/visualization/visualization_config_functions.cc#L217) in master to something like this:

```
result.publish_period = config.publish_period * 32.0;
```

With the [default publish_period](https://github.com/RobotLocomotion/drake/blob/38d6dce4734fa596b1c0db5515e57bb2569f4cd2/visualization/visualization_config.h#L45) = 1/64 seconds (frequency 64 Hz) in VisualizationConfig, we'll publish contact at 2 Hz. The Gamepad response time seems reasonable, although we can see a visual delay of contact surfaces following moving objects around.  It's not bad if I move the paddle slowly.

You can play with my branch at https://github.com/DamrongGuoy/drake/tree/clutter_sim_DG_2022-06-01

DamrongGuoy, 2023-06-05:
I think we can unblock students for now.  I have a working branch at:

- https://github.com/DamrongGuoy/drake/tree/clutter_sim_DG_2023-06-05

It added:
1. `self._meshcat.Flush()` for flow controls between the producer (clutter_sim) and the consumer (meshcat).
2. `cylinder_collision_medium.obj` for a collision geometry that has half triangles of the visual geometry.
3. Print the number of triangles of each of the 16+ contact surfaces. Originally we got around 360 triangles per contact patch. With the `_collision_medium.obj`, we got around 170 triangles per contact patch, and MeshCat was happier.

I'll delete the other branches now.

DamrongGuoy, 2023-06-07:
@joemasterjohn @SeanCurtis-TRI and I had a meeting for a longer-term plan to improve MeshCat contact visualization.  The meeting note is [here](https://docs.google.com/document/d/10lA_VQ6Mb1IwmXlhlSMSTfwDEiDhdxOXE0j4digyhLU/edit?usp=sharing) (it's TRI document, but I can grant access).

DamrongGuoy, 2023-06-08:
@jwnimmer-tri suggested an idea that, for visualization, we filter for contacts with the actuated paddle, so we draw only a few contact patches without the other 16 contact patches between free bodies. I also confirmed with @rcory that it is a good idea to try.

Toward that goal, I have this working branch [clutter_sim_DG_2023-06-07](https://github.com/DamrongGuoy/drake/tree/clutter_sim_DG_2023-06-07).  It has a small LeafSystem that gets ContactResults from MultibodyPlant and passes the results to ContactVisualizer. However, it lacks Python bindings for modifying ContactResults. Right now it can only report the number of contacts with the paddle before passing all contacts to ContactVisualizer.

I can also see this simple LeafSystem as useful for logging contact results with the paddle for the Sim2Real study.

DamrongGuoy, 2023-06-08:
For experiments, I added `ContactResults::Select(GeometryId)` to create a copy of ContactResults with only hydro contact info with a specific GeometryId.

MeshCat can draw only contacts with the paddle.  Now I can get around 100% rtr most of the time.  Users can focus their attention on the paddle without distraction from free-body contacts (cylinders, walls, table top).

See my branch [clutter_sim_DG_2023-06-08](https://github.com/DamrongGuoy/drake/tree/clutter_sim_DG_2023-06-08).  I delete the 2023-06-07 branch.

https://github.com/RobotLocomotion/drake/assets/42557859/5a0257a9-70d9-4525-a045-5559e9f37e30

rcory, 2023-06-15:
Closing this issue, since my immediate problem has been resolved by following @DamrongGuoy's recommendations (see his update to the original issue description). Also #19592 and #19593 capture features that will further help mitigate this issue.

closed 2023-06-15 by rcory