RobotLocomotion/drake #19544: Unable to parse thin rod via SDFormat

jwnimmer-tri, 2023-06-06:
### What happened?

The SDFormat parser fails to parse a valid link inertia.

This demo program:
```python
from pydrake.all import *

def main():
    model = """
<?xml version="1.0"?>
<sdf version="1.7">
  <model name="semiacrobot">
    <link name="link1">
      <inertial>
        <pose>0 20 0 0 0 0</pose>
        <mass>1</mass>
        <inertia>
          <ixx>0.083</ixx>
          <iyy>0.083</iyy>
          <izz>0</izz>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyz>0</iyz>
        </inertia>
      </inertial>
    </link>
  </model>
</sdf>
    """
    plant = MultibodyPlant(0)
    Parser(plant).AddModelsFromString(model, ".sdf")

main()
```

... produces this output ...

```
WARNING:drake:<literal-string>.sdf:5: warning: Spatial inertia fails SpatialInertia::IsPhysicallyValid().
 mass = 1.0
 Center of mass = [0.0  20.0  0.0]
 Inertia about point P, I_BP =
[400.083        0        0]
[      0    0.083        0]
[      0        0      400]
 Inertia about center of mass, I_BBcm =
[0.083      0      0]
[    0  0.083      0]
[    0      0      0]
 Principal moments of inertia about Bcm (center of mass) =
[0.0  0.083  0.08300000000002683]
```

... and substitutes a stand-in spherical inertia for the actual thin rod inertia.

The desired outcome is no warnings, and the actual inertia remains intact as given by the user (within a few epsilon, I suppose).

I am not sure whether the parser code is misusing the Multibody API (i.e., a parser bug), or whether the Multibody API has precision-matching problems, or something else.

### Version

v1.17.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

jwnimmer-tri, 2023-06-06:
For the record, we discovered this while developing the user-facing test cases for #19210.

mitiguy, 2023-06-06:
The following code reproduced the error.
Note: Currently this code is in spatial_inertia_test.cc for a possibly pending PR.
Below is my first guess as to what/why this is happening.
```
  // Create a spatial inertia for a thin rod B about its center of mass Bcm.
  const double mass = 10.0, length = 5.0;
  const SpatialInertia<double> M_BBcm_B =
      SpatialInertia<double>::ThinRodWithMass(mass, length, Vector3d(0, 0, 1));

  // Form M_BBo_B (B's spatial inertia about a point Bo, expressed in B) using 
  // the mass, position from Bo to Bcm, and B's rotational inertia about Bcm.
  // Note: With mass = 10 kg and the distance-squared from Bo to Bcm as
  // dcm_squared = 10² + 20² + 30² = 1400, mass * d_cm_squred = 14000 kg*m².
  // This 14000 kg*m² is much larger than rod B's maximum moment of inertia
  // about Bcm of 1/12 * mass * length² ≈ 20.8333 (ratio 14000 / 28.8333 = 672).
  // Since numbers of these differing size are added to form M_BBo_B and then
  // subtracted to shift back to Bcm for the validity check, it seems reasonable 
  // that digits are lost in the process -- and it would be helpful to account
  // for these addition/subtraction tolerance problems in the validity check.
  const RotationalInertia<double> I_BBcm_B = M_BBcm_B.CalcRotationalInertia();
  const Vector3<double> p_BoBcm_B(10, 20, 30);
  const SpatialInertia<double> M_BBo_B =
     SpatialInertia<double>::MakeFromCentralInertia(mass, p_BoBcm_B, I_BBcm_B);
```

mitiguy, 2023-06-07:
PR #19551 helps resolves issue #19544.

PR #19581 (2023-06-14): SpatialInertia::MakeFromCentralInertia() immediately checks physical validity
Closes #19544

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19581)
closed 2023-06-14 by mitiguy