RobotLocomotion/drake #19564: Add Staging Docker job

BetsyMcPhail, 2023-06-08:
Currently, the [push_release](https://github.com/RobotLocomotion/drake/blob/master/tools/release_engineering/dev/push_release) script pulls the nightly Docker images for Focal and Jammy from Docker Hub, tags them with the appropriate platform prefix and the version number, and then pushes the tag back to Docker Hub.

This process complicates the `push_release` script by requiring a date. Also, the docker images really should be created from the result of the staging jobs (e.g. https://drake-jenkins.csail.mit.edu/view/Staging/job/linux-focal-unprovisioned-gcc-bazel-staging-packaging/).

This issue proposes creating a docker staging job that will create the appropriate docker images, tag them and push them to Docker Hub. 

Note that the nightly packaging jobs (e.g. https://drake-jenkins.csail.mit.edu/view/Packaging/job/linux-focal-unprovisioned-gcc-bazel-nightly-packaging/) already push to Docker Hub.

BetsyMcPhail, 2023-06-08:
@mwoehlke-kitware, @jwnimmer-tri I noticed that the Staging Packaging jobs already create docker images, they just don't push them. Would it be possible (from a policy not technical point of view) to update that job instead of creating a new Docker-specific job?

mwoehlke-kitware, 2023-06-19:
Strictly speaking, this change as written only involves changes on the drake-ci side, but since those do nothing until we actually make use of them, it probably makes sense for #19631 to be the closing PR.

Mostly, though, it should be noted that the drake-ci changes happened in RobotLocomotion/drake-ci#227.

PR #19631 (2023-06-27): Rewrite push_release logic for docker
Rewrite logic to push the release docker tags in Python, and to rely on the staging tags we are newly creating. This hopefully makes the logic easier to follow, and allows for a dry-run. More importantly, it eliminates the need to associate a release with a nightly build, which we really shouldn't have been doing in the first place, and simplifies things for the release manager.

Also remove an outdated instruction from the README, and change the aptly snapshot to use the source version rather than the corresponding nightly date for the release's snapshot  [... cut]