RobotLocomotion/drake #19580: Support robot-relative IK in Differential Inverse Kinematics Integrator

marip8, 2023-06-12:
**Is your feature request related to a problem? Please describe.**
Currently the `DifferentialInverseKinematicsIntegrator` class [expects the input target poses to be relative to the fixed root frame of the multibody plant](https://github.com/RobotLocomotion/drake/blob/37d6c59a13431956d82123de6cda24cfa765c50d/multibody/inverse_kinematics/differential_inverse_kinematics.cc#LL251C1-L256C34) (as far as I can tell). In a lot of robotics applications, target poses are specified relative to a robot frame (e.g., from a camera on the robot, etc.), and this robot frame may not coincide directly with the fixed world frame.

With the merge of #19522 per [this post](https://stackoverflow.com/questions/76391333/differential-inverse-kinematics-integrator-does-not-work-with-a-floating-plant), this class will fully support floating base configurations. For an application that I am working on, my robot floats in space and its target poses are specified in robot coordinates (i.e., by a camera attached to the robot); but since the base is floating and all of my operations are robot-relative, my simulation isn't portable to reality if I reason about transforms with respect to a fixed world frame that I won't actually be able to know.

I would like for Drake to solve IK in the root frame of a multibody plant model, rather than the fixed world frame, such that I could eventually deploy my Drake-based code on a physical system (i.e. without the Drake simulator back-end) and have it still work without knowledge of a fixed world frame. It seems like this is not currently possible because solving differential IK relies on [transforming poses to the fixed world frame](https://github.com/RobotLocomotion/drake/blob/37d6c59a13431956d82123de6cda24cfa765c50d/multibody/inverse_kinematics/differential_inverse_kinematics.cc#L251-L253).

For these cases described, I think the simulations would correlate more directly with reality if the input target poses were assumed to be in the root frame of a specific model of the multibody plant. It would also be convenient (but not totally necessary) if the user could also specify an arbitrary frame to which the target poses were relative and let Drake handle the necessary transformations to get the pose into the "correct" frame internally for IK.

**Describe the solution you'd like**
1. Add a constructor for the relevant IK classes that takes a model and calculates IK for the joints in `model` assuming the poses are relative to the root frame of the model, rather than the fixed world frame
    ```python
    DifferentialInverseKinematicsIntegrator(plant, model, tip_frame, ...)
    ```
1. Add a constructor for the `DifferentialInverseKinematicsIntegrator` that takes a model and pose reference frame name string and calculates IK for the joints in `model` assuming the poses are relative to `ref_frame`
    ```python
    DifferentialInverseKinematicsIntegrator(plant, model, ref_frame, tip_frame, ...)
    ```
1. Change `DoDifferentialInverseKinematics` to look up the transform of the target pose with respect to a specified reference frame, which should not be required to be the fixed world frame

**Describe alternatives you've considered**
Users could transform their robot-relative poses to the fixed world frame themselves and then connect to the `DifferentialInverseKinematicsIntegrator`. This seems a bit inconvenient and bakes in the assumption that the real-life application will also know about a fixed world frame, which is not true in some floating-base applications.

The target pose input port to the `DifferentialInverseKinematicsIntegrator` could also be changed to a `Frame` type that contains both a `RigidTransform` and a string reference frame name, but that seems like a less backwards compatible change.

**Additional context**
I have only used the Python version of Drake, but I'm happy to help contribute to a PR with these changes if it's helpful

RussTedrake, 2023-06-21:
I think it would be very reasonable to offer a reference frame as an additional overload for DoDifferentialInverseKinematics.  I'll try to finish up https://github.com/RobotLocomotion/drake/pull/19522 in the next day or two (sorry I got sidetracked!), then will revisit this.

RussTedrake, 2023-06-30:
#19522 landed the other day.  Sorry for that delay.  I think it would be great to follow-up with this feature.

My sense is that the first step should be to add a new entry point in `differential_inverse_kinematics.h`.  We currently have
```
DifferentialInverseKinematicsResult DoDifferentialInverseKinematics(
    const MultibodyPlant<double>& plant,
    const systems::Context<double>& context,
    const Vector6<double>& V_WE_desired,
    const Frame<double>& frame_E,
    const DifferentialInverseKinematicsParameters& parameters)
``` and ```
DifferentialInverseKinematicsResult DoDifferentialInverseKinematics(
    const MultibodyPlant<double>& robot,
    const systems::Context<double>& context,
    const math::RigidTransform<double>& X_WE_desired,
    const Frame<double>& frame_E,
    const DifferentialInverseKinematicsParameters& parameters);
```
which, as you say, artificially imposes the world frame.  (`DifferentialInverseKinematicsIntegrator` currently uses the second entry point, which delegates back to the first).  These methods then call [`CalcJacobianSpatialVelocity`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a5f39a8f68fae3de53767a281503a3313), which take the measured in (and expressed in) frames; `DoDifferentialInverseKinematics` is just forcing them to the world frame.

I think we could expose the richer form of the Jacobian API.  What do you think about adding the following two new methods? First:
```
DifferentialInverseKinematicsResult DoDifferentialInverseKinematics(
    const MultibodyPlant<double>& plant,
    const systems::Context<double>& context,
    const Frame<T>& frame_B,
    const Eigen::Ref<const Vector3<T>>& p_BP,
    const Frame<T>& frame_A,
    const DifferentialInverseKinematicsParameters& parameters)
```
This provide the relevant part of the `CalcJacobianSpatialVelocity` API, including your requested case. I think in our DiffIK implementation, we always want the Jacobian wrt V and the expressed in frame (`frame_E`) equal to the measured in (`frame_A`).  The second new method would be the analogous form that takes `X_AB_desired` as an input.  The existing two methods would just delegate to these implementations; I don't see any reason to deprecate the existing APIs which can be seen as convenient short-cuts.

My recommendation would be that we land these changes first, with the unit tests, and then we can make the second small update to reroute `DifferentialInverseKinematicsIntegrator` to use the new version?  If you're willing to try PR'ing that, we'd be happy to help you get it through!

PR #20973 (2024-02-21): Support relative frames in DifferentialInverseKinematicsIntegrator
Adds support for passing a relative frame A into the constructor. Deprecates the X_WE_desired input port in favor of the more general X_AE_desired input port.

Resolves #19580.

+@iantheengineer for feature review, please?

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20973)
closed 2024-02-21 by RussTedrake