RobotLocomotion/drake #19582: Discuss MinimumValueConstraint formulation

hongkai-dai, 2023-06-13:
Currently given a vector `v`, MinimumValueConstraint https://github.com/RobotLocomotion/drake/blob/master/solvers/minimum_value_constraint.h tries to enforce that the min(v) >= v_min.

I am trying to achieve two tasks in this github issue
1. Understand the current mathematical formulation that enforces min(v) >= v_min.
2. Propose the mathematical formulation to enforce min(v) <= v_min_ub.

## Current formulation for min(v) >= v_min
In our current formulation, we enforce this through
```
smooth_max( φ((vᵢ - v_influence)/(v_influence - vₘᵢₙ)) / φ(-1) ) ≤ 1
```
where `v_influence` is an upper bound of all `v`, φ(x) is a penalty function (like smoothed hinge loss) that typically approximates the function max(-x, 0). `smooth_max` is a smoothed max function that over approximates `max` (namely smooth_max(v) >= max(v)). Also smooth_max(x) works nicely if x is within [-1, 0], hence we want the the argument to `smooth_max` to be within this range [-1, 0].

I don't understand why we introduce the penalty function φ here. So far as I can see, we could enforce the following constraint
```
smooth_max((v_influence - vᵢ)/(v_influence - vₘᵢₙ)) <= 1
```
which would also be a smoothed version of min(v) >= v_min. Here is a quick proof
```
 smooth_max((v_influence - vᵢ)/(v_influence - vₘᵢₙ)) <= 1 
⇒ max((v_influence - vᵢ)/(v_influence - vₘᵢₙ)) <= 1 
⇔ max(v_influence - vᵢ) <= v_influence - vₘᵢₙ
⇔ max(- vᵢ) <=  - vₘᵢₙ
⇔ min(vᵢ) >=  vₘᵢₙ
```
So why we decide to introduce the penalty function φ?

@calderpg-tri @avalenzu

rpoyner-tri, 2023-06-13:
Assigning @hongkai-dai as component owner for further delegation.

hongkai-dai, 2023-06-22:
OK, I think I understand the original design now. i will put my understanding here for future reference.

The reason why we use 
```
smooth_max( φ((vᵢ - v_influence)/(v_influence - vₘᵢₙ)) / φ(-1) ) ≤ 1
```
instead of
```
smooth_max((v_influence - vᵢ)/(v_influence - vₘᵢₙ)) <= 1 
```
is to handle the discontinuity of gradient when certain vᵢ crosses v_influence.

We assume that `v` only include values below `v_influence` (for example in our collision checking algorithm, we just ignore the pair if the distance is above v_influence). This sudden change on the number of entries in `v` would cause discontinuity in the value of smooth_max(v). To remedy this, we pad `v` with `v_influence` to keep the size of `v` fixed. But for these padded entries in `v`, we also need to set its gradient. Without further information, we can only set the gradient of the padded entries to 0. This would cause an issue because when the entry value is below v_influence, its gradient is non-zero, which would lead to discontinuity in gradient when the value crosses `v_influence`. To handle this discontinuity in the gradient, we use the `φ(x)` function, which has zero gradient when `x=0`.

I will close this issue as it is resolved.

closed 2023-06-22 by hongkai-dai