RobotLocomotion/drake #19586: Check Context Validity when Getting Collision Filter Manager

cohnt, 2023-06-13:
### What happened?

It appears that the method `scene_graph.collision_filter_manager(context)` does not perform any checking of the validity of the context it is given. See, for instance, the following code:
```
from pydrake.all import DiagramBuilder, AddMultibodyPlantSceneGraph

builder = DiagramBuilder()
plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step=0.0)
diagram = builder.Build()
plant.Finalize()

context = diagram.CreateDefaultContext()
plant_context = plant.GetMyContextFromRoot(context)
sg_context = scene_graph.GetMyContextFromRoot(context)

filter_manager = scene_graph.collision_filter_manager(sg_context) # Correct
filter_manager = scene_graph.collision_filter_manager(context) # Should throw an error, but does not
filter_manager = scene_graph.collision_filter_manager(plant_context) # Segfaults
```
As in the comments, both the scene graph context and root contexts allow it to proceed without error. I would think passing in the root context would cause an error. (I have had issues where passing in the root context does lead to an error, but I couldn't come up with a concise MWE.) Passing in the multibody plant's context led to a segmentation fault.

cohnt, 2023-06-13:
Here's a backtrace of the segmentation fault:
```
#0  0x00007fff4e90551b in drake::systems::AbstractValues::get_mutable_value(int) () from /home/tommy/opt/rlg/drake-build/install/lib/python3.10/site-packages/pydrake/common/../../../../libdrake.so
#1  0x00007fff4f1441f7 in drake::geometry::SceneGraph<double>::mutable_geometry_state(drake::systems::Context<double>*) const () from /home/tommy/opt/rlg/drake-build/install/lib/python3.10/site-packages/pydrake/common/../../../../libdrake.so
#2  0x00007fff4f144982 in drake::geometry::SceneGraph<double>::collision_filter_manager(drake::systems::Context<double>*) const () from /home/tommy/opt/rlg/drake-build/install/lib/python3.10/site-packages/pydrake/common/../../../../libdrake.so
#3  0x00007fff41b5fc38 in ?? () from /home/tommy/opt/rlg/drake-build/install/lib/python3.10/site-packages/pydrake/geometry/__init__.cpython-310-x86_64-linux-gnu.so
#4  0x00007fff41a0378f in ?? () from /home/tommy/opt/rlg/drake-build/install/lib/python3.10/site-packages/pydrake/geometry/__init__.cpython-310-x86_64-linux-gnu.so
#5  0x00005555556b0c9e in cfunction_call (func=<built-in method collision_filter_manager of PyCapsule object at remote 0x7fff4416f240>, args=<optimized out>, kwargs=<optimized out>) at ../Objects/methodobject.c:543
#6  0x00005555556a772b in _PyObject_MakeTpCall (tstate=0x555555b5c380, callable=<built-in method collision_filter_manager of PyCapsule object at remote 0x7fff4416f240>, args=<optimized out>, nargs=<optimized out>, keywords=0x0) at ../Objects/call.c:215
#7  0x00005555556bf1eb in _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=0x7ffff752a3a8, callable=<built-in method collision_filter_manager of PyCapsule object at remote 0x7fff4416f240>, tstate=0x555555b5c380) at ../Include/cpython/abstract.h:112
#8  _PyObject_VectorcallTstate (kwnames=<optimized out>, nargsf=<optimized out>, args=0x7ffff752a3a8, callable=<built-in method collision_filter_manager of PyCapsule object at remote 0x7fff4416f240>, tstate=0x555555b5c380) at ../Include/cpython/abstract.h:99
#9  method_vectorcall (method=<optimized out>, args=0x7ffff752a3b0, nargsf=<optimized out>, kwnames=<optimized out>) at ../Objects/classobject.c:53
#10 0x000055555569f75a in _PyObject_VectorcallTstate (kwnames=0x0, nargsf=<optimized out>, args=0x7ffff752a3b0, callable=<method at remote 0x7fff3ed62840>, tstate=0x555555b5c380) at ../Include/cpython/abstract.h:114
#11 PyObject_Vectorcall (kwnames=0x0, nargsf=<optimized out>, args=0x7ffff752a3b0, callable=<method at remote 0x7fff3ed62840>) at ../Include/cpython/abstract.h:123
#12 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, trace_info=0x7fffffffd4b0, tstate=<optimized out>) at ../Python/ceval.c:5891
#13 _PyEval_EvalFrameDefault (tstate=<optimized out>, f=<optimized out>, throwflag=<optimized out>) at ../Python/ceval.c:4181
#14 0x0000555555696176 in _PyEval_EvalFrame (throwflag=0, f=Frame 0x7ffff752a240, for file /home/tommy/tmp.py, line 14, in <module> (), tstate=0x555555b5c380) at ../Include/internal/pycore_ceval.h:46
#15 _PyEval_Vector (tstate=0x555555b5c380, con=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>) at ../Python/ceval.c:5065
#16 0x000055555578bc56 in PyEval_EvalCode (co=<code at remote 0x7ffff74adb00>, 
    globals={'__name__': '__main__', '__doc__': None, '__package__': None, '__loader__': <SourceFileLoader(name='__main__', path='/home/tommy/tmp.py') at remote 0x7ffff73ddba0>, '__spec__': None, '__annotations__': {}, '__builtins__': <module at remote 0x7ffff75905e0>, '__file__': '/home/tommy/tmp.py', '__cached__': None, 'DiagramBuilder': <pybind11_type(__init__=<instancemethod at remote 0x7fff51ad99c0>, __doc__="DiagramBuilder is a factory class for Diagram.\n\nIt is single use: after calling Build or BuildInto, DiagramBuilder\ngives up ownership of the constituent systems, and should therefore be\ndiscarded; all member functions will throw an exception after this\npoint.\n\nWhen a Diagram (or DiagramBuilder) that owns systems is destroyed, the\nsystems will be destroyed in the reverse of the order they were added.\n\nA system must be added to the DiagramBuilder with AddSystem or\nAddNamedSystem before it can be wired up in any way. Every system must\nhave a unique, non-empty name.\n\nBuilding large Diagrams\n---------...(truncated), locals=<optimized out>) at ../Python/ceval.c:1134
#17 0x00005555557b8b18 in run_eval_code_obj (tstate=0x555555b5c380, co=0x7ffff74adb00, 
    globals={'__name__': '__main__', '__doc__': None, '__package__': None, '__loader__': <SourceFileLoader(name='__main__', path='/home/tommy/tmp.py') at remote 0x7ffff73ddba0>, '__spec__': None, '__annotations__': {}, '__builtins__': <module at remote 0x7ffff75905e0>, '__file__': '/home/tommy/tmp.py', '__cached__': None, 'DiagramBuilder': <pybind11_type(__init__=<instancemethod at remote 0x7fff51ad99c0>, __doc__="DiagramBuilder is a factory class for Diagram.\n\nIt is single use: after calling Build or BuildInto, DiagramBuilder\ngives up ownership of the constituent systems, and should therefore be\ndiscarded; all member functions will throw an exception after this\npoint.\n\nWhen a Diagram (or DiagramBuilder) that owns systems is destroyed, the\nsystems will be destroyed in the reverse of the order they were added.\n\nA system must be added to the DiagramBuilder with AddSystem or\nAddNamedSystem before it can be wired up in any way. Every sy [... cut ...]
#26 0x000055555577ebcd in Py_BytesMain (argc=<optimized out>, argv=<optimized out>) at ../Modules/main.c:720
#27 0x00007ffff7c29d90 in __libc_start_call_main (main=main@entry=0x55555577eb90 <main>, argc=argc@entry=2, argv=argv@entry=0x7fffffffd968) at ../sysdeps/nptl/libc_start_call_main.h:58
#28 0x00007ffff7c29e40 in __libc_start_main_impl (main=0x55555577eb90 <main>, argc=2, argv=0x7fffffffd968, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd958) at ../csu/libc-start.c:392
#29 0x000055555577eac5 in _start ()
```

jwnimmer-tri, 2023-06-13:
I think the fix is that the `SceneGraph<T>::mutable_geometry_state(Context<T>* context)` implementation needs to call `this->ValidateContext(context);` prior to using the context.

Probably also `SceneGraph<T>::geometry_state(const Context<T>& context) const` should likewise gain the same check.

See also #11252 for related work.

closed 2023-06-14 by jwnimmer-tri