RobotLocomotion/drake #19592: [multibody] Select only contact results of a specified geometry

DamrongGuoy, 2023-06-14:
**Is your feature request related to a problem? Please describe.**
Related to
- #19494 

For visualization, in #19494 we found a use case to both accelerate MeshCat and also help users to focus on contact patches of the actuator and to skip drawing contact patches among free bodies.  In that example, instead of drawing 16 table-cylinder contact patches, we would draw only a few paddle-cylinder contact patches. A [prototype](https://github.com/RobotLocomotion/drake/compare/master...DamrongGuoy:drake:clutter_sim_DG_2023-06-08) showed that Gamepad response time becomes almost real-time.

**Describe the solution you'd like**
In [multibody/plant/contact_results.h](https://github.com/RobotLocomotion/drake/compare/master...DamrongGuoy:drake:clutter_sim_DG_2023-06-08#diff-c237799815d208d42eef1fe795c2981d011b7bc5edb30b167999c65d7801635b), provide this advanced API

```C++
  /** (Advanced) Experimental API to select all hydroelastic contacts with a
   specific collision geometry. It enables users to process only contact info
   of a manipulator and skip contact info between free bodies.
   See issue [19494](https://github.com/RobotLocomotion/drake/issues/19494). */
  ContactResults Select(geometry::GeometryId selector) const;
```
With that API, users can make a simple system to select desirable contact patches for visualization and also downstream processing (e.g., sim2real verification) [like this.](https://github.com/DamrongGuoy/drake/blob/b62403f4fceed9bda2a88f40e39ef4b469a0b7c1/examples/clutter/clutter_teleop_sim.py#L145-L179)

**Describe alternatives you've considered**
Expose the four [hidden APIs ](https://github.com/RobotLocomotion/drake/blob/9e49f35e858e372d7a5f4c58c7419febc2381599/multibody/plant/contact_results.h#L63-L81) of ContactResults to public.  However, I think that would be overkill.

**Beneficiary**

@rcory @rthomasson23 of TRI and Stanford.

**Additional context**
Add any other context or screenshots about the feature request here.  The following movie shows only the contact patches on the paddle as it moves around. Users can focus on the paddle contacts without distraction from other contacts between free bodies.

https://github.com/RobotLocomotion/drake/assets/42557859/6421c7bb-ab4d-480d-9397-2e5f78e45ed9

jwnimmer-tri, 2023-06-14:
> `ContactResults Select(geometry::GeometryId selector) const;`

This is too narrow.  It suits your use case, but what about users who want to see _two_ geometries' contacts?

One example of a more general form would be a "remove contact" function like this:

```c++
  /** Removes the ith PointPairContactInfo instance. The input index i
   must be in the range [0, `num_point_pair_contacts()` - 1] or this method
   aborts. Higher-numbered contact indices are shifted down by 1. */
  void RemovePointPairContactInfo(int i);

  /** Removes the ith HydroelasticContactInfo instance. The input index i
   must be in the range [0, `num_hydroelastic_contacts()` - 1] or this
   method aborts. Higher-numbered contact indices are shifted down by 1. */
  void RemoveHydroelasticContactInfo(int i);
```

Alternatively, the functions could take a `std::vector<int>` indices.

Alternatively, the functions could take predicate callback `std::function<bool(const HydroelasticContactInfo<T>&)> should_remove`.

DamrongGuoy, 2023-06-14:
I like the idea to be more general by predicate callback `std::function<bool(const HydroelasticContactInfo<T>&)>.

I wonder how it would work with Python.  Would C++ callback to Python predicate slow things down significantly or not?  I'm making a prototype [branch](https://github.com/DamrongGuoy/drake/tree/select_contact_results) to test it.

DamrongGuoy, 2023-06-14:
I have tested my prototype branch [select_contact_results](https://github.com/DamrongGuoy/drake/tree/select_contact_results) in this python test branch [test_clutter_sim_select_contact](https://github.com/DamrongGuoy/drake/tree/test_clutter_sim_select_contact).  The response time looks fine to me.

I'll continue with the idea to use a predicate callback.

closed 2023-06-27 by jwnimmer-tri

DamrongGuoy, 2023-06-27:
#19663 is in master now.  The new function is [ContactResults::SelectHydroelastic()](https://github.com/RobotLocomotion/drake/blob/90e0ad2bb2e1c76d45d6d9c1115b1207e2528f7b/multibody/plant/contact_results.h#L61-L70).

I have updated my branch [test_clutter_sim_select_contact](https://github.com/DamrongGuoy/drake/tree/test_clutter_sim_select_contact) to use `SelectHydroelastic()` in [this block](https://github.com/DamrongGuoy/drake/blob/ea9d52d8785324c374d218ad1a54699ab903dfd3/examples/clutter/clutter_teleop_sim.py#L173-L177) of code.  The Gamepad response time looks good to me.

@rcory when you have time, could you try it out, please?

PR #19663 (2023-06-27): [multibody] Select a subset of hydroelastic ContactResults
Fix #19592

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19663)