RobotLocomotion/drake #19599: Modeling of conveyor belts

amcastro-tri, 2023-06-14:
While we could in Drake model the entire mechanics of a conveyor belt, it is not computationally practical, introducing additional DoFs and complicating the setup.

Imagine a conveyor belt of a finite length L, width W and height H.
We propose to model the conveyor as:
 1. Box of size L x W x H, anchored to the world, introducing no state.
 2. We assign a (new) `ProximityProperty`, lets call it "surface velocity vₛ" (a vector) for now. As with coulomb friction, assigning this property to a given geometry, affects the surface of that geometry.
 3. The meaning of this velocity is: "the surface somehow slips around the object at velocity vₛ, like if the object had a jacket we pull off at velocity vₛ"
 4. Contact computations would include this bias term, effectively modeling a surface that is moving at velocity vs, even though the geometry is anchored.

This feature would allow us to model conveyor belts of finite length, without complicating geometry or adding state, and it'd also allow us to build more complex setups of belts made up of parts at different orientations in the plane and  vertical slopes.

SeanCurtis-TRI, 2023-06-14:
Several questions:

1. The surface velocity is defined in the body's frame?
2. How coupled to the idea of being a box should it be? What if the property is applied to spheres or other arbitrary shapes?
3. You used the phrase "slips __*around*__ the object" (emphasis mine). So, does that suggest you expect the velocity on the top of the box is in the opposite direction as that of the bottom?
    - How would you articulate "bottom"? How about the sides where the velocity has to transition from "forwards" to "backwards" as we transition from "top" to "bottom"?
    - See my proposal below.
4. Does the conveyor belt have to be anchored?

Proposal:

For a body with arbitrary geometry, we define a plane in the body's frame and a conveyor speed. The most parameter is the direction of the plane normal (reversing the normal will reverse the direction of the conveyor belt). See the image below.

To compute the surface velocity at a point on the geometry we:

  1. Compute the normal at the point.
  2. Project the normal onto the plane.
  3. Rotate the vector around the plane's normal. This vector now points in the direction of the surface velocity.
  4. Scale the velocity vector to match the desired conveyor speed.
 
[image]

amcastro-tri, 2023-06-15:
FYI, Per conversation with Jeremy, "... we would need this implemented [...] by early/mid July ..."
So simpler is better.

jwnimmer-tri, 2023-06-15:
For actual wheels, you'd probably need a more ergonomic way to adjust the speed.  Updating proximity property values with a controller isn't really practical, it'd need to be an input port or something.

amcastro-tri, 2023-06-15:
Excellent observation @jwnimmer-tri. 
That'd require an input put somewhere. Do you have an idea where? what that API could look like? 
Even if we do not support that on a first pass, I believe I'd like to at least not write code that hinder us from doing this on a second pass. I believe the `ProximityProperty` approach I was proposing would hinder us from accomplishing this.

It might just be all an MBP API with full control on the bodies (and their geometries) being affected along with possible new input ports.

jwnimmer-tri, 2023-06-15:
For me at least, it's hard to generalize without working through some examples.  I would suggest to outline each of your use cases along with pseudo-code for how the user would set up and simulate their system(s).

For a port API, one option would be `MultibodyPlant::DeclareBodySurfaceSpeedInputPort` to provide opt-in to this feature.  It could take arguments for the BodyIndex as well as the details of the plane (storing those as Parameters), and then the input port is just the 1-dimensional speed.

Another thing to consider is whether this applies to bodies or geometries.  As a plant feature, I suppose bodies makes the most sense, in which case the speed is mapped onto all proximity geometry for that Body?  If the user wants one logical body where only some geometry has a surface speed, they could split their logical body into two links that are welded together, where only one of the links has a surface speed.

jwnimmer-tri, 2023-06-15:
Also consider: is it 1-dimensional speed, or actually velocity?  I could imagine wanting to run backwards.

jwnimmer-tri, 2023-06-15:
Also consider: even if you allow for the input port approach, it might make sense to still offer a proximity property, to handle the common case where the speed is a constant.  That would allow users to have their SDFormat model files hard-code the speed, without extra software setup steps.  Perhaps the plane parameter should also be in the SDFormat file, instead of (or in addition to) the MbP function argument(s).  Working through some sample uses cases will probably help you resolve these kinds of questions.

jbinagia, 2023-06-15:
@jwnimmer-tri @amcastro-tri I just wanted to add one additional comment/request which is that, as you can imagine from the real life scenario, we'd like to be able to adjust the speed of the conveyor dynamically during simulation (e.g. turning the conveyor on and off). So the speed would need to be able to be set/modified after calling `Finalize()` for the plant. Y'all may have already assumed this was the case, but I figured I would call it out just to be clear

sherm1, 2023-06-15:
Rather than a port, would a Parameter be sufficient for setting the speed? That seems simpler if adequate.

SeanCurtis-TRI, 2023-06-16:
I took for granted the idea that they'd have a controller for the speed in the diagram. @jbinagia Indicated the desire to "adjust the speed of the conveyor __*dynamically during simulation*__" (emphasis mine). That can be interpreted either way. @jbinagia do you want to clarify?

Furthermore, if we're hoping to use this for caterpillar wheels, then the port seems absolutely essential.

jwnimmer-tri, 2023-06-16:
I suspect at some point we're going to want the framework to by-default offer the ability to wire up any parameter(s) as input ports, without needing code in individual leaf systems to allow that one by one.  We've frequently heard requests to adjust parameters during a simulation, using diagram wiring.  For now though, I think a dedicated port makes sense out of expediency.

rpoyner-tri, 2023-06-16:
Geometry question about how well does the geometry definition generalize: can we define surface velocity on a sphere (or stranger shapes) this way without problems? Could we (say) model a ping pong ball with spin?

I don't know that any problems from the above question block the work, but it might be helpful to think about generalizing when naming the new software elements.

Feel free to ignore me -- I'm heading to vacation soon, and wanted to leave that thought here for later.

jbinagia, 2023-06-16:
@amcastro-tri I like this proposal - I think it  meets all of our needs. I'll put some thoughts below in case they help to clarify any of the discussion (feel free to ignore if irrelevant)
* I should clarify we don't absolutely need to model a caterpillar wheel per se. For example, I think it is sufficient for our use to even just have a box geometry that has a velocity applied at its top surface (so in effect, modeling the motion of the top of the conveyor but not modeling the entire belt itself). I figured I would call this out in case it makes the solution easier. But, If the best solution encompasses both the caterpillar wheel as well as a simple box geometry then I say we should go ahead with that option for the sake of generality/future-proofing
* @SeanCurtis-TRI Would it be possible to cover both cases (setting speed with/without a controller attached)? I suppose this would be achieved with the input port. Because I can imagine scenarios where one wants to set the conveyor speed with a controller, but I can also imagine a scenario where you just want some very simple logic to set the conveyor speed inside the program, e.g. `conveyor_speed = 1.0 if conveyor_turned_on else 0.0`

amcastro-tri, 2023-06-16:
@rpoyner-tri Nothing precludes us from using this model on any abitrary geometry. Of course "a sphere with a belt" makes no sense. However, the modeler could use this feature to impose spin on a wheel without having to model the dynamics of the wheel. If we ever allow the unit vector b̂ , it'd allow the modeling of things like an omni wheel.

@jbinagia Yes, the conveyor idea is no more complex than a single value, so I'd say lest do the belt model.

@jbinagia recall everything in Drake follows the system's framework abstraction. What you propose above (`conveyor_speed = 1.0 if conveyor_turned_on else 0.0`) is nothing but a simple custom `LeafSystem`. Information can only flow through ports.

jbinagia, 2023-06-16:
@amcastro-tri A clarification - are you still considering defining this as a `ProximityProperty` as @jwnimmer-tri mentioned [here](https://github.com/RobotLocomotion/drake/issues/19599#issuecomment-1593577933)? And `MultibodyPlant::DeclareBeltSurfaceVelocity(const Body<T>& body, const Vector3<T>& b, const T& default_belt_speed);` would override any values defined in the SDF/URDF?

SeanCurtis-TRI, 2023-06-16:
@jbinagia I'd second what @amcastro-tri said. The word "controller" may have been misleading. I'd suggest that your `conveyor_speed = 1.0 if conveyr_turned_on else 0` *is* a controller. In fact, the interesting question is not what the speed gets set to, but what the circumstances are that would cause `conveyor_turned_on` to toggle.

So, you've got two issues:

1. You've got some logic somewhere that flips the on/off switch.
2. Based on the switch, you need to feed a speed to the conveyor.

In this case, the meat of that is where the switch logic lives. You could have an observer outside the simulation that determines that the switch needs to flip. Alternatively, you could have a `LeafSystem` inside the diagram that observes state in the `Diagram` and has an output port that indicates switch state.

The nice thing about @amcastro-tri's proposal is that either approach would work. The system in the diagram would be slightly easier (I believe).

For an external observer, you could simply fix the speed value on the input port. When the observer logic detects that the switch needs to be flipped, you'd stop the simulation, and fix the opposite value on the port. The hard part here is *stopping* the simulation.

For a `LeafSystem`, the switching would happen exactly when the system's output port logic decides to flip, so there's no stopping the simulation. You just have to make sure it has access to the appropriate state to inform the switch logic via its own input port.

Either way, this will get the job done.

scpeters, 2023-06-17:
For reference, Open Dynamics Engine models "conveyor belt" surface motion for a friction pyramid model with `dContactMotion1`, `dContactMotion2`, and `dContactMotionN` parameters

* http://ode.org/wiki/index.php?title=Manual#Contact

SeanCurtis-TRI, 2023-06-21:
I believe you can think of it as an actually moving object (instead of an abstract surface with velocity). The contact force computation (and dynamic response) cannot distinguish *why* the relative velocity at a point is what it is, only that the relative velocity exists. So, if things could slip with a moving box. It should be able to slip just as easily with a "moving" surface.

juanmed, 2025-07-10:
@jwnimmer-tri @amcastro-tri  I will be giving this a try. I am new to the drake codebase, just started with the tutorials last week. Any pointers on how to attack the implementation?

amcastro-tri, 2025-07-16:
I like how you piped the surface speed @juanmed. It looks like you are tyring to use the continuous computation of contact forces? for simulation, we typically use our discrete time stepping strategy in the SAP solver. Are you familiar with that?

Essentially, for this to work with SAP, we'd need to add a bias term in our contact constraints, so that the contact velocities are now vc = J⋅v+b, with J the contact Jacobian, v the generalized velocities and b the new bias term (essentially the belt's velocity expressed in the contact frame). 

@jbinagia asked for this same feature today in Drake. If you guys would find it useful, I could either guide you trhough the math or write a prototype and if you guys like it, you can help me pushing it as a full Drake feature with parsing and pythoon bidings? how does that sound? We can talk more on Slack if that's easier.

juanmed, 2025-07-22:
@amcastro-tri Thanks for the comments, I was looking today at all those functions and its starting to make sense.

If you have time, would you mind taking a look at how I am computing normals [here](https://github.com/RobotLocomotion/drake/pull/23197/files#r2219775080) and how I added the surface velocity to the separation velocity [here](https://github.com/RobotLocomotion/drake/pull/23197/files#r2219821781)? I know it is still unrelated to SAP, wust want to make sure I am not wasting effort on the wrong direction. Thank you.

jwnimmer-tri, 2025-07-22:
@amcastro-tri 

>  (feel free to Slack us on the Drake Developers!)

Our slack channel is private and invitation-only, and costs is money for every person we invite, and so we only invite long-term collaborators into it.  Generally, pull requests should be discussed in the pull request interface, not in slack.

jwnimmer-tri, 2026-04-16:
My initial thought is that we should have 1 input port always provided, and use a `BusValue` as its type.  Then the only question is what names to use for the topics on the `BusValue`.  Probably `Body::scoped_name()`?  I would need to be convinced that we need shape-level control over velocities.

closed 2026-08-24 by SeanCurtis-TRI